mavlink/qgroundcontrol #10885: How to load qgroundcontrol android repo on android stuido

Malikfarooq, 2023-11-30:
I want to load the android repo  of qgroundcontrol project into my Android studio but I am unable to load this its showing  like that 
[image] 
 please help me where I am wrong

DonLakeFlyer, 2023-11-30:
> please help me where I am wrong

You can't use android studio. You need to use Qt Creator.

closed 2023-11-30 by DonLakeFlyer

bhaskar-glitch, 2023-12-01:
> > please help me where I am wrong
> 
> You can't use android studio. You need to use Qt Creator.

We are willing to add some other functionality in the app. Like adding a dashboard to display udp data. Can we do that using QT creator?? and further will we able to export it as a application??

DonLakeFlyer, 2023-12-01:
https://dev.qgroundcontrol.com/master/en/getting_started/