mavlink/qgroundcontrol #10887: QGC Daily: Moved to Qt version 6.6.1 - Status

DonLakeFlyer, 2023-12-02:
How you can help:
* Try out a daily build and report issues you find making sure to label the Issue with the "Qt6" label.
* Recommendation is to try your scenarios in SITL first before using QGC Daily on a real vehicle flight
* Join the discussion on the QGC discord channel for questions. Don't clutter GitHub Issues with questions.

Current status:
* Windows, Linux and Mac builds are available from the normal Daily Build links on qgrouncontrol.com
* You are likely to run into quirkiness/bugs in the UI. It is going to take some time to find all the missing changes to the Qml UI code in QGC. 
* The required changes to Qml although mostly mechanical in nature are not 100% discoverable through just code inspection. It's going to take running all the parts of the ui and verifying it works correctly to find the remaining needed changes.
* The C++ code should be very close to 100% updated. Those issues show up at compile time as opposed to run time.
   * The largest changes were to the Map plugin which download tiles. So there may still be problems to wring out there. 
* The majority of the top level UI with respect to Vehicle Setup and flight of PX4 and ArduPilot vehicles has been tested to verify the basics are working correctly.
* Plan View has had less focus and id likely to be more flaky than the rest of QGC.

Known Issues:
* Video is currently disabled in all OS types. Video support is still a WIP.
* Android builds are not yet available. Android is still WIP.
* Log download is not yet available

Building from master:
* Docs are not yet updated. That will come in the future.
* Qt version you need is 6.6.1
* CI builds are still using qmake for build. So that is known to be in a working state. Transition to make for CI will start soonish.
* Here is a screen shot of the modules needed from Mac OSX. You should be able to figure what you need in your OS from here:
[image]

zdanek, 2023-12-02:
@DonLakeFlyer Bluetooth is hidden in Connectivity

DonLakeFlyer, 2023-12-02:
Add info on what needs to happen to get android build working

jstrebel, 2023-12-06:
Link to daily build is broken

XMoonRiverX, 2023-12-06:
I tried android build using windows but got error

[image]
[image]

DonLakeFlyer, 2023-12-20:
FYI: I'm working on getting the android build running now

DonLakeFlyer, 2023-12-23:
Android is building now. Still needs more testing to find problems.

julianoes, 2024-01-15:
Has there been a change how the unit tests can be run? If I add `--unittest` nothing happens.

DonLakeFlyer, 2024-01-16:
Nope. It could be busted though. I'll look.

DonLakeFlyer, 2024-01-16:
It's working fine for me on master

julianoes, 2024-01-16:
Is there a build flag that I might be missing?

DonLakeFlyer, 2024-01-16:
Only trigger for unit tests is debug build

julianoes, 2024-01-16:
Aha, that might be the issue. Thanks.

Kim-zhi-jiang, 2024-01-21:
Looking forward the map plugin run.
Qt6.6.1 so much change, this problem "qt.network.ssl: No functional TLS backend was found"  copy "stl" Qt ../plugins/stl  into project plugins then can be work

julianoes, 2024-01-21:
@Kim-zhi-jiang 

> copy "stl" Qt ../plugins/stl into project plugins then can be work

Can you explain with more detail what we have to do? I don't quite understand. Are there docs about it somewhere?

Kim-zhi-jiang, 2024-01-22:
> @Kim-zhi-jiang
> 
> > copy "stl" Qt ../plugins/stl into project plugins then can be work
> 
> Can you explain with more detail what we have to do? I don't quite understand. Are there docs about it somewhere?

[Xcb, X11, Xlib, Wayland](https://venam.net/blog/unix/2016/10/25/x11-blah-blah.html)
Map plugin not work (Ubuntu 22.04 Qt 6.6.1 X11)  
I'm a rookie, but I know it's really difficult to adapt to so much.

Kim-zhi-jiang, 2024-01-22:
> @Kim-zhi-jiang
> 
> > copy "stl" Qt ../plugins/stl into project plugins then can be work
> 
> Can you explain with more detail what we have to do? I don't quite understand. Are there docs about it somewhere?

I try install openssl1.1.1 in to Ubuntu 22.04 but failed.
Hope to fix Qt 6.6. x in the next version 
[How to build Qt 5.15 Community with OpenSSL 3](https://gist.github.com/seyedmmousavi/b1f6681eb37f3edbe3dcabb6e89c5d43)

julianoes, 2024-01-23:
My assumption was that Qt6 would come with OpenSSL 3 but does it still not?

Kim-zhi-jiang, 2024-01-23:
> My assumption was that Qt6 would come with OpenSSL 3 but does it still not?

I also tried build the openssl 3 source code in the Qt6.6.1 tools.
The map is still blank .

DonLakeFlyer, 2024-01-23:
I've run on Ubuntu 22 and the map worked fine. I haven't seen map problems on any OS.

DonLakeFlyer, 2024-01-23:
New video support was just merge into daily builds. Woohoo!

jinglebang, 2024-05-16:
why  I  build  master source code  in the Qt6.6.3   is  Daily version

jinglebang, 2024-05-16:
[image]
[image]

closed 2024-11-30 by HTRamsey