mavlink/qgroundcontrol #10891: Video is not recorded using the suspension ViewLink Q30Trim pro

Ed7son, 2023-12-05:
When I try to start a video recording, nothing happens. When searching for a problem, I found that the video manger = nill

zdanek, 2023-12-05:
@Ed7son how to reproduce this?
If you could add more info, as specified in bug template, we could have a chance to reproduce and fix this problem.

Ed7son, 2023-12-12:
I have this logs. But I forgot the version because my version is very changed and the version number is no longer displayed
[QGC VIDEO LOG.md](https://github.com/mavlink/qgroundcontrol/files/13644660/QGC.VIDEO.LOG.md)

Ed7son, 2023-12-12:
> @Ed7son how to reproduce this? If you could add more info, as specified in bug template, we could have a chance to reproduce and fix this problem.

- Operating System: Linux (ubuntu 22.04.3 LTS)
- QGC Version: 4. (forgot because my version very changed and the version number is no longer displayed)
- QGC build: daily
- Autopilot - Ardupilot copter 4.3.8

[QGC VIDEO LOG.md](https://github.com/mavlink/qgroundcontrol/files/13644755/QGC.VIDEO.LOG.md)

closed 2026-10-04 by github-actions[bot]