mavlink/qgroundcontrol #10914: QGC Parameters changed are not being reflected

Jackwooh, 2023-12-08:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
IMU should be disabled and RAW IMU data should show zero values. (In MAVLink Inspector
[image]
[image]
)

## Current Behavior
IMU are still running even after disabling them.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Load ardupilot firmware file (stable one), from https://firmware.ardupilot.org/Plane/
2. Upload the frimware from QGC gui.
3. Change the INS_USE, INS_USE1, INS_USE2 parameters to disable
4. It should give error that any one IMU should be available but the IMU are still running.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 14
- QGC Version: 4.3.0
- QGC build: stable
- Flight Controller: Pixhawk 2
- Autopilot (with version): Ardupilot Plane 4.4.3

## Detailed Description
Change in parameters are not being reflected. I disabled all the parameters of IMU but the sensors are still working, it was working fine 2 weeks ago, but suddenly it stopped working, i want to disable certain IMU's but it doesn't seem to work.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2023-12-18:
I'm pretty sure you need to reboot the autopilot for these changes to take effect. ArduPilot should have marked these parameters as reboot required but that would be a firmware bug.

QGC does not do anything other than chnage the parameter values. From there on out firmware is in control and what should  happen.

closed 2023-12-18 by DonLakeFlyer

Jackwooh, 2023-12-26:
@DonLakeFlyer I rebooted the QGC but the parameters are not changing, later i discovered that the problem is on linux its running good on windows.

DonLakeFlyer, 2023-12-26:
> I rebooted the QGC but the parameters are not changing

You need to reboot the vehicle not QGC