mavlink/qgroundcontrol #10938: MPEGTS KLV synchronization

WingRS, 2023-12-26:
Hi,
I've noticed that QGroundControl supports MPEGTS, which is fantastic. 
I've also seen some PR on object tracking selection GUI, which is great as many developers who work with QGround are building their tools for object tracking, and it would be great to have standard support for this in QGround. 
But, there is a hidden problem with object selection for tracking - timestamps. Drones tend to fly at some speed, and objects can also be moving; thus, sending bounding box coordinates with timestamps is crucial.
A standard way to do this is by using KLV synchronization that comes with MPEGTS  - [here is a page from ridge run](https://developer.ridgerun.com/wiki/index.php/GStreamer_In-Band_Metadata_for_MPEG_Transport_Stream).

So a working scheme for this would be
1. Target selection in GUI
2. Target selected
3. KLV - timestamp of the frame fetched from the KLV
4. MAVLINK message generated - bounding box and timestamp
5. Message sent  

This would be great to have either for PX4 and Ardupilot, and it could be standard [MAVLINK Command Long](https://mavlink.io/en/messages/common.html#COMMAND_LONG),  with message type [Tracking Rectangle](https://mavlink.io/en/messages/common.html#CAMERA_CAP_FLAGS_HAS_TRACKING_RECTANGLE), so it would have the timestamp and the bounding box data. 

I think if QGround will have it in its release it will stop developers inventing the GUIs on and on and this will be a commodity for the research, DIY, industry in future. 

Thanks,
Yurii

zdanek, 2024-01-24:
It's very interesting. Currently I'm involved in stabilizing video stream in new Qt6 based version, but the proposal seems interesting.

WingRS, 2024-02-13:
I've jsut checked and now target tracking selection GUI is also merged, so this would be an amazing addition!

> It's very interesting. Currently I'm involved in stabilizing video stream in new Qt6 based version, but the proposal seems interesting.