mavlink/qgroundcontrol #10943: 3D map viewer new feature for QGC

omid-esrafilian, 2023-12-29:
I have recently added a 3D viewer into QGC to visualize the 3D map of the environment, the 3D model of the vehicle (multirotor), and the mission 3D trajectory (including the waypoints). The 3D map for any region of interest can be downloaded from the OpenStreetMap website (.osm file) and imported to this 3D viewer. You can see a demonstration of this new feature from the link below (YouTube video).
I have also attached a screenshot of this viewer integrated into QGC.

Link to the video:
https://youtu.be/aJI5349crG8?si=DUIGgef6J9bnKjxN

The viewer works for any flight control that is compatible with GQC and it supports multiple vehicles as well.
Currently, only a 3D model for multirotors is added to this 3D viewer. However, this can be easily extended for other types of vehicles.
The Qt version that I used for this project is 5.15.2. But it can also be migrated to Qt 6.

Please let me know if I can create a new pull request to merge the code. The current branch that I am working on is Stable_V4.3.

Thanks,

[image]

DonLakeFlyer, 2024-01-04:
Thanks for explaining what you are working on. 

My first question is: What is the use case for this? It certainly cool, but what user problem does it solve?

The reason I ask:
* Any new top level feature in QGC can clutter the UI making it harder to use. So if some new thing doesn't have good reasons to be added then it best to leave things simpler.
* Also any large pack of new code which doesn't get much usage tends to get orphaned. In that it's left to the maintainers to keep it up to date and working when the original developer is gone.

omid-esrafilian, 2024-01-05:
Dear @DonLakeFlyer ,

Thank you for your remarks. 

I can imagine/envision several use cases for this new feature:

- While the QGC is more suited for outdoor settings where GPS is available, there are many scenarios where vehicles need to operate in indoor environments. In these cases, this 3D viewer can potentially enable users to visualize the vehicle location and the 3D map of the environment (e.g. the point cloud) constructed by the robot/vehicle in real-time.

- In the case of simulation, for instance, in SITL or HITL modes, the simulation results, such as the status and movements of the vehicle, can be shown in the QGC (instead of using Gazebo or other similar applications).

- Another use case would be navigation in an environment by exploiting the 3D map data. Since the 3D map is made available by this viewer, this information can be used to facilitate the navigation (i.e. path planning) of the vehicle, for instance, to avoid collisions with objects and obstacles in an (outdoor) environment. This means, expanding the QGC capabilities towards a centralized flight management unit of vehicles for safer and collision-free operations. However, this use case would be for more advanced applications.

Regarding the maintenance and development, I believe that others would also be interested in this new feature, as I have had several people reach out to me expressing interest in this 3D viewer. I am confident that there will be more people willing to contribute to this specific part. 
We have also already used this new feature in one of our projects. I included below the link to the project for your reference:
https://www.youtube.com/watch?v=y4Ffqh7eoFQ&t=17s

Please let me know if I need to provide further information on this new feature.

Thanks,

DonLakeFlyer, 2024-01-05:
Ok, great thanks for the info. Next questions:

Is the 3D viewer based on a separate third party library. If so what is the licensing on that? And what is the cross-platform support for it?

omid-esrafilian, 2024-01-05:
@DonLakeFlyer 

I have used a third-party library called "Earcut". The license is "ISC License". It basically grants for any purpose. I have included the link to the "Earcut" project below for your reference. Other than that, It is mainly Qt.
https://github.com/mapbox/earcut.hpp/tree/master

I have built and run the code successfully on Windows and Linux (Ubuntu). I didn't have the setup ready for Android, iOS, or Mac OS. So, it has to be tested for those platforms.

DonLakeFlyer, 2024-01-05:
Ok, I think we are good to go from that standpoint then. Once you move to master/Qt6 when don't you submit a pull. I'd like to put the UI to turn this on someplace other than where you have it. But it's fine for now as a starting point.

omid-esrafilian, 2024-01-05:
@DonLakeFlyer 
Ok, great. I will start working on the branch master and migrating the code to Qt6.
Once it's finished, I'll let you know.

Thanks

omid-esrafilian, 2024-01-06:
@DonLakeFlyer 

I have moved to the master/Qt6 branch. So, I will submit a pull request as you suggested.

closed 2024-05-31 by HTRamsey

iura1, 2024-07-19:
Hi @omid-esrafilian ,

Great feature!
Is there any way I can see the terrain also in 3d (the hills and mountains)? Currently I see in 3d just the buildings, is this working with your current implementation from the master branch with .osm file? 

Thanks a lot!

omid-esrafilian, 2024-07-19:
> Hi @omid-esrafilian ,
> 
> Great feature! Is there any way I can see the terrain also in 3d (the hills and mountains)? Currently I see in 3d just the buildings, is this working with your current implementation from the master branch with .osm file?
> 
> Thanks a lot!

@iura1 
Thanks for your interest.
There have been some upgrades to this feature. I have added satellite imagery along with the 3D building imported from the OSM file. The terrain data has not been added yet. Unfortunately, I have been busy lately, so I could not add more features. Hopefully, soon I can add more features. Please note that this feature is only available under the **master** branch.
Please check the link below for more details on the latest changes and how to activate the 3D viewer:
https://docs.qgroundcontrol.com/master/en/qgc-user-guide/viewer_3d/viewer_3d.html