mavlink/qgroundcontrol #10951: Action buttons disappeared in APK build from latest version 4.3.0

orca741, 2024-01-09:
After several attempts, I finally managed to build the latest version (4.3.0) for Android OS. In official  docs for developers, it is mentioned that version 5.12 must be used, but actually it must be 6.6 or more and really building the source for Android was very challenging. including the `not found` error for many `.qml` files and etc. anyway
The current problem is that the action buttons are not displayed in the `firstrunprompt`.
[image]

DonLakeFlyer, 2024-01-10:
That is on purpose. If the action button is only Ok or Close then they are no longer shown. You just click outside to close the dialog. Or Escape/Enter key. I may put them back if this seems too strange. But at this point this is working correctly.

orca741, 2024-01-10:
Ok!
But this is android device and there is no escape/enter key. Also clicking outside not working in this prompt at least.
Please fix that!

DonLakeFlyer, 2024-01-11:
Use a daily build downloaded from website. Verify that also doesn't work for you. It's woking fine for me. If daily build created from CI works, then there is something wrong with your own personal build.

closed 2024-06-03 by HTRamsey