mavlink/qgroundcontrol #10973: Radio Not Ready fault during first set up (+can't fit mag 1 fault)

Nosgnod5754, 2024-01-16:
Hey everybody, I'm pretty new to this stuff. I'm connecting a RadioMaster Pocket to a Pixhawk 2.4.8 clone through an RP3 RadioMaster receiver. The devices are all charged, and the radio is bound to the receiver. But when I go to QGroundControl to set up the radio, every time I press Calibrate, I get a message saying, "Radio Not Ready, Please turn on transmitter" (attached below). I know that the radio is connected because when I unplug the Pixhawk, the radio says, "Telemetry lost". So what should I do now?

Also, when setting up the Compass, I keep failing because of a "can't fit mag 1" fault (see attached). I suspect that either the compass in the Pixhawk or the GPS module is faulty. Not sure if this is related to the issue above though. Any advice would be appreciated. Thank you. 

[image]

[image]

umangpatel4, 2024-03-26:
hii have you solved this problem i am facing similar one

huimingao-finder, 2024-08-06:
I also meet the same question!!! Waiting for answer...

HM778, 2024-08-06:
also waiting for answer...

Nosgnod5754, 2024-08-06:
> also waiting for answer...
> hii have you solved this problem i am facing similar one

Hey guys, so turns out it was a communication issue between my receiver and my flight controller. Turns out, Spektrum and Pixhawk stuff don't mix too well for some reason. At least I haven't had very much luck with them. ELRS stuff didn't work either for some reason. I did try connecting my ELRS receivers to UART ports on the Pixhawk but no luck. But then I found out that I can configure my Radiomaster receivers to communicate with my flight controller using Sbus instead, so I wired it up for SBUS, reconfigured it in the handheld controller's settings, and it took right off!!!! Here's the link, good luck.
https://oscarliang.com/how-to-output-sbus-from-an-expresslrs-receiver/

Nosgnod5754, 2024-08-06:
> also waiting for answer...
> hii have you solved this problem i am facing similar one
Looks like this method would work for most ELRS receivers, not just radiomaster. Oh and make sure you have a transmitter that can bind with your ELRS receiver as well. If that doesn't work for you, make sure your receiver and transmitter are bound correctly, make sure your receiver gets the voltage that it needs (some ports on the pixhawk only give out 3.3 V while some receivers need 5V), make sure your solders are good (check for high ohms with a multimeter). If all else fails, switch computers with somebody else to see if that's the issue. 

About the mag 1 thing, turns out it's because I was calibrating it indoors. you gotta calibrate magnetometers outdoors or else it will get really confused by the steel reinforcements inside your walls.

HM778, 2024-08-07:
Thanks for your immediate reply !!!   I've checked connections and ensured they are correct. 
I finally realized there is a configure mistake in "src/drivers/rc_input/RCInput.cpp", it should be "rc_io_invert(**false**)" in "case RC_SCAN_SBUS:"(line 453) instead of "rc_io_invert(true)" , because my hardware has invert the signal. 
Hope our answers could be helpful to others!

HM778, 2024-08-07:
> Hey guys, so turns out it was a communication issue between my receiver and my flight controller. Turns out, Spektrum and Pixhawk stuff don't mix too well for some reason. At least I haven't had [...]

Thanks for your immediate reply !!! I've checked connections and ensured they are correct. I finally realized there is a configure mistake in "src/drivers/rc_input/RCInput.cpp", it should be "rc_io_invert(**false**)" in "case RC_SCAN_SBUS:"(line 453) instead of "rc_io_invert(true)" , because my hardware has invert the signal. Hope our answers could be helpful to others!

closed 2026-10-04 by github-actions[bot]