mavlink/qgroundcontrol #10986: PID window scaling 5 inch screens

Siciro, 2024-01-21:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issue
## Expected Behavior
PID window scaling correctly with 5 inch screen

## Current Behavior
Window scaling sometimes to small or too big and x and y axis figures cut off. 

## Steps to Reproduce:
Refer to attached photod

## System Information
Siyi radio mk 15 with 5 inch screen

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
-
[image]

[image]
[image]

Siciro, 2024-01-21:
Found this error in the velocity window
[image]
[image]

DonLakeFlyer, 2024-01-23:
I believe that manual tuning which is popping those missing parameter error is only available on latest dev firmware

closed 2024-02-26 by DonLakeFlyer

Siciro, 2024-03-01:
> New px4 stable should be coming in a couple months. Leaving this alone for
> now

Hello Don.

Will that be 1.15?