mavlink/qgroundcontrol #10994: QGC on ubuntu cannot connect to the network

Bruch-wang, 2024-01-24:
## Expected Behavior
QGC can not connect the network the map cannot be loaded

## Current Behavior
----------------------------------------------------

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.QGC version:v4.3.0 ubuntu version:20.04
2.QGC can not connect the network the map cannot be loaded
3.we try the different QGC version such as v2.7.0 and so on but still have this problem
4.we suspect that the map api on ununtu has expired or is experiencing some issues that event changing the map source won't fix

## System Information
QGC v4.3.0   Ubuntu 20.04

## Detailed Description
--------------------------------------------------------

## Log Files and Screenshots
--------------------------------------------------------

zdanek, 2024-01-24:
Did you try other map providers?
You can change it in settings.
Enable logs and see if there are errors.

Bruch-wang, 2024-01-25:
Yes. I have tried different map sources before, such as Google and Bing map sources, but still can not load the map, and I noticed in the console log display is not connected to the Internet, I think maybe this is not the map source problem but QGC itself network connection problem, and this is the issue in ubuntu20.04, I'm running everything fine on Windows

Bruch-wang, 2024-01-25:
[image]

[image]

Kim-zhi-jiang, 2024-01-31:
Internet problem!!! 
I also repeat this problem,  (I accidental touch  once can able to load  maps in short time,this time github no need tunnel )

yang5757, 2024-03-29:
me too, The console log says Fetch tile error Connection closed.my system information is window11, the QGroundControl is 4.3.0
[image]
[image]

closed 2024-11-30 by HTRamsey