mavlink/qgroundcontrol #10998: Qt6.6.1 Android compile error: No rule to make target AndroidManifest.xml on windows

osenberg-x, 2024-01-26:
## Expected Behavior
QGC can be compiled successfully.

## Current Behavior
[image]

## Steps to Reproduce:
follow official document: [official document link](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/)
[image]

## System Information
When posting bug reports, include the following information
- Operating System: [Windows11]
- QGC Version: [master]
- QGC build: [self-built from source]
- Qt version: [Qt6.6.1]

## Detailed Description
```

17:21:03: "D:\Qt\6.6.1\android_armv7\bin\qmake.bat" D:\ws\qgroundcontrol\qgroundcontrol.pro -spec android-clang "CONFIG+=debug" "CONFIG+=qml_debug"

'which' is not recognized as an internal or external command, operable program or batch file.

Project MESSAGE: ANDROID_TARGET_ARCH armeabi-v7a armeabi-v7a
Project MESSAGE: Qt version 6.6.1
Project MESSAGE: Android Arm 32 bit build
Project MESSAGE: Daily Build
Project MESSAGE: GIT_DESCRIBE v4.3.0-124-g8b479120f
Project MESSAGE: QGroundControl APP_VERSION_STR VERSION v4.3.0-124-g8b479120f 4.3.0.124
Project MESSAGE: VERSION 4.3.0.124
Project MESSAGE: Android version info: 344300124 bitness:34 major:4 minor:3 patch:00 dev:124
Project MESSAGE: Debug flavor
Project MESSAGE: Using MAVLink dialect 'all'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Including support for video streaming
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
17:21:06: 进程"D:\Qt\6.6.1\android_armv7\bin\qmake.bat"正常退出。
17:21:06: 正在启动 "E:\env\Android\Sdk\ndk\25.1.8937393\prebuilt\windows-x86_64\bin\make.exe" -f D:/ws/build-qgroundcontrol-Qt_6_6_1_Clang_armeabi_v7a-Debug/Makefile qmake_all

make: Nothing to be done for 'qmake_all'.
17:21:07: 进程"E:\env\Android\Sdk\ndk\25.1.8937393\prebuilt\windows-x86_64\bin\make.exe"正常退出。
17:21:07: 正在启动 "E:\env\Android\Sdk\ndk\25.1.8937393\prebuilt\windows-x86_64\bin\make.exe" -j16

make: *** No rule to make target 'D:\ws\build-qgroundcontrol-Qt_6_6_1_Clang_armeabi_v7a-Debug\ANDROID_PACKAGE_SOURCE_DIR\AndroidManifest.xml', needed by 'libQGroundControl_armeabi-v7a.so'.  Stop.
17:21:07: 进程"E:\env\Android\Sdk\ndk\25.1.8937393\prebuilt\windows-x86_64\bin\make.exe"退出，退出代码 2 。
Error while building/deploying project qgroundcontrol (kit: 安卓 Qt 6.6.1 Clang armeabi-v7a)
When executing step "Make"
```

andrefreitas97, 2024-02-15:
I have the same problem :( Did you manage to solve it?

anhhien2k, 2024-03-01:
i have the same problem. Do you have a solution for this problem?

ZHCLN, 2024-03-06:
i have the same problem. Do you have a solution for this problem?

ZHCLN, 2024-03-11:
> I have the same problem :( Did you manage to solve it?

Have you solved it now?

closed 2024-04-29 by HTRamsey