mavlink/qgroundcontrol #11004: Regarding Customization of Ui of Qground Control

arunba1, 2024-01-31:
Hi All,

I’m pretty new to the drone scene, so please forgive any newbie mistakes I might (actually, will definitely) make.

I wanted to customize the UI of QGroundControl. I do not have any real idea where to start. I read the developer pages on their website here. Unfortunately, I was not able to make heads nor tails of where I am to start for this task. I’m using Qt 6.6.1 and windows 11 as OS . I’m not able start , as I don’t know where the qml files are located which are required for UI customization. Please help me out

closed 2024-01-31 by DonLakeFlyer

DonLakeFlyer, 2024-02-06:
https://discord.com/channels/1022170275984457759/1022185820683255908

arunba1, 2024-02-06:
Sir, please verify this link as I'm not to join sir.