mavlink/qgroundcontrol #11013: QGCS Custom Build Failed

Storch9120, 2024-02-01:
----
## Timeline
I’ve followed the latest instructions for a custom build i.e.

1. Fresh Build (completes)
2. Renamed custom-example to custom
3. Ran the python script updateqrc.py
4. Clean build
5. Rebuild (fails)
    and encountered the following errors (Re: Screenshot)
[image]

So I made these changes to the CustomFirmwarePlugin.cc at line 23 (inside if statement)
[image]

Now the build completes but on execution this is shown in Application Output:
[image]

## System Information
- Operating System: Ubuntu 20.04
- QGC build: Daily 
- Qt Version 6.6.1

Please advise if I should revert those changes I made and what other changes I should make to diagnose these errors.

DonLakeFlyer, 2024-02-02:
The custom build hasn't been updated to move to Qt6 yet. I'll try to get to it as soon as I can.

ijulanday, 2024-02-07:
@Storch9120 hit the same issues! Fix for me was:
* make the same changes as you did to CustomFirmwarePlugin.cc 
* add `import Qt5Compat.GraphicalEffects` to CustomAttitudeWidget.qml (the last line of your final error message)
The application loads and I can switch between fly & plan view, all the buttons seem to work as expected. Not much rigorous testing I can do as I'm just getting into this stuff myself.

[image]
[image]

TishSerg, 2024-04-09:
> The custom build hasn't been updated to move to Qt6 yet. I'll try to get to it as soon as I can.

Any update on this?

HTRamsey, 2024-04-13:
#11013

closed 2024-04-13 by HTRamsey