mavlink/qgroundcontrol #11015: Android App crashing while downloading parameters after connecting in custom build

dilipsai1, 2024-02-02:
Android App crashing while downloading parameters after connecting in custom build
Custom build was built on Stable version 4.3
----

## Expected Behavior
Cube orange should connect successfully
## Current Behavior
App  built with stable version 4.3,app connects to cube orange,shows downloading and crashes..have attached log file and image for reference 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.Build custom app using Stable version 4.3
2.Build completes and generates apk
3.install app in mobile
4.connect orange cube to app
5.app crashes

## System Information
When posting bug reports, include the following information
- Operating System: Windows 11
- QGC Version: Stable 4.3
- QGC build: Stable
- Flight Controller: Pixhawk cube orange
- Autopilot (with version): Ardupilot

## Detailed Description
App was  built successfully with stable version 4.3,app connects to cube orange over bluetooth,shows downloading and crashes..have attached log file and image for reference 

## Log Files and Screenshots
[Log_4.3.txt](https://github.com/mavlink/qgroundcontrol/files/14135547/Log_4.3.txt)

[image]

DonLakeFlyer, 2024-02-02:
Have you verified that regular QGC Stable 4.3 crashes as well? Or is this just a problem with your custom build?

dilipsai1, 2024-02-06:
> Have you verified that regular QGC Stable 4.3 crashes as well? Or is this just a problem with your custom build? Also give regular daily build a try. Trying to narrow down where the problem is.

Stable build fails with below error:

make: *** No rule to make target 'C:\Users\Wer\QGC_Normal_4.3\build-qgroundcontrol-Android_Qt_5_15_2_Clang_Multi_Abi-Debug\ANDROID_PACKAGE_SOURCE_DIR\AndroidManifest.xml', needed by 'libQGroundControl_x86.so'.

Could you please help @DonLakeFlyer

DonLakeFlyer, 2024-02-06:
Don't try building yourself just test the built version.

closed 2026-10-04 by github-actions[bot]