mavlink/qgroundcontrol #11017: Throttle with joystick is only 50% so makes it unable to calibrate

flyasky, 2024-02-03:
## Expected Behavior
Able to calibrate joystick.

## Current Behavior
Unable to proceed with calibration because QGC sees just 50% of the axis when I move the throttle fully down.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. joystick is Radiomaster zorro controller with EdgeTX latest, model with classic joystick and AETR mixes applied. (it could be calibrated in DRL, Lift-Off with this setup).
2. no settings in ~/.mavlink/joysticks
3. run SITL + QGC
4. load module joystick in mavproxy
5. open QGC > Vehicle Setup/ Joystict Setup, try to calibrate (see attached images)

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04
- QGC Version: 4.3.0
- QGC build: stable, appimage
- Flight Controller: just simulation
- Autopilot (with version): Ardupilot Copter 4.5

## Detailed Description

pic1 (maximum, left stick is fully up, see axis2/4)
[image]

pic2 (minimum, left stick is fully down, see axis2/4)
[image]

mtzm, 2024-03-27:
I experience the same issue with the FrSky Taranis.

After finding this and this [issue](https://github.com/mavlink/qgroundcontrol/issues/10104), the clues lead to the Game Controller interface of SDL2. After running [jstest-sdl](https://github.com/meleu/jstest-sdl.git) i noticed that the joystick inputs are correct when running `jstest-sdl --test 0`, but wrong when running `jstest-sdl --gamecontroller 0`.
Running `jstest-sdl -l` showed following game controler mapping:
`Mapping: '030068cd09120000544f000011010000,OpenTX FrSky Taranis Joystick,a:b0,b:b1,x:b3,y:b4,back:b10,guide:b12,start:b11,leftstick:b13,rightstick:b14,leftshoulder:b6,rightshoulder:b7,leftx:a0,lefty:a1,rightx:**a3**,righty:**a4**,lefttrigger:**a2**,righttrigger:a5,crc:cd68,platform:Linux'`

Running QGroundControl with the channels rearranged like below solved the issue for me:
`SDL_GAMECONTROLLERCONFIG="030068cd09120000544f000011010000,OpenTX FrSky Taranis Joystick,a:b0,b:b1,x:b3,y:b4,back:b10,guide:b12,start:b11,leftstick:b13,rightstick:b14,leftshoulder:b6,rightshoulder:b7,leftx:a0,lefty:a1,rightx:**a2**,righty:**a3**,lefttrigger:**a4**,righttrigger:a5,crc:cd68,platform:Linux" ./QGroundControl.AppImage`

saito1111, 2025-11-04:
hi same issue , i forked https://github.com/JacopoPan/aerial-autonomy-stack/ and i modified the docker command to mount  
/dev/input  https://github.com/JacopoPan/aerial-autonomy-stack/blob/main/scripts/docker/Dockerfile.ground 
When i calibrate my controller (radiomaster zorro ) i am blocked on the "push all the throttle down" step .
When i do it only goes to zero on qgc , like i am only able to move from 0 (which correspond to my max down position) and +100 (which correspond to my max up position )

kende0211, 2025-11-24:
> Unclear as to whether anyone has tried this on 5.0? Can someone let me know if this repros on 5.0? Anything older than 5.0 is no longer supported.

I have built a few days ago from the git master branch (5.0.8).
Mtzm's fix is okay for me, but it could be better, and I still consider it as an issue.

DonLakeFlyer, 2026-01-22:
Joystick has been reworked in daily builds. You should try that first. The fixes in. daily are not going to go back into stable.

M4rkoHR, 2026-09-02:
Still a problem with RadioMaster Pocket on v5.1.4 Ubuntu 26.04 LTS despite jstest showing throttle -32k to +32k just like all other channels. How do you fuck up a CALIBRATION procedure, if 0 is minimum then it's the minimum, don't blame SDL

DonLakeFlyer, 2026-09-02:
@M4rkoHR If you don't like the support you are getting then you are welcome to contribute a pull yourself to fix it. Either that or I'll gladly refund your purchase price for QHC.

donpurnell, 2026-09-08:
I have the same problem with EdgeTX radio controller RadioMaster TX16S, USB connection to Ubuntu 24.04.

@[[Mtzm's fix]( https://github.com/mtzm) ]   does solve the problem, though qGroundControl  v5.1.4  displays a weird re-mapping of channels, that I cannot explain.

For the  TX16S I can run from a terminal:
=================================
export SDL_GAMECONTROLLERCONFIG="<my TX16S UUID>,RadioMaster TX16S,a:b0,b:b1,leftx:a0,lefty:a2,rightx:a3,righty:a1,"

./QGroundControl-x86_64.AppImage
=================================
I extracted <my TX16S UUID>  from
cat /proc/bus/input/devices | grep -E "Name|Sysfs|Handlers"
by converting 4 Raw Hex numbers   to  4 Little-Endian numbers, and concatenating.

 It seems unreasonably difficult, if  an OpenTX  controller on Ubuntu is a moderately common  setup.