mavlink/qgroundcontrol #11020: Video streaming seems not work in windows qt6 qgc

hyeonseung89, 2024-02-06:
## Expected Behavior
Want to RTSP Video source play and work at qt6 QGC
test source is rtsp://210.99.70.120:1935/live/cctv004.stream

## Current Behavior
Recently saw that gstreamer fixes for qt6 qgc were merged.  But it doesn't seem to play in Windows apps. I haven't tried any other OS.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior

1. Download generated latest app by CI action
2. In video settings, set source to rtsp 
3. input address

## System Information
When posting bug reports, include the following information
- Operating System: Windows 11 Home 22H2
- Graphic Card: rtx3060ti
- QGC Version: master branch
- QGC build: daily
- gstreamer version : 1.18.1
- Qt 6.6.1 msvc 2019 64bit

## Detailed Description
When I compiled it myself up to qt 5.15, it worked normally.

## Log Files and Screenshots
- Screenshots of QGC console.
[image]

DonLakeFlyer, 2024-02-06:
@zdanek Do you have the ability to test under Windows?

zdanek, 2024-02-06:
Yes I do.

DonLakeFlyer, 2024-02-06:
Thank so much for looking at this.

AlexKlimaj, 2024-02-09:
```[!] at D:\a\qgroundcontrol\qgroundcontrol\src\VideoReceiver\GstVideoReceiver.cc:1370 - "GStreamer error: Could not initialize window system"```

DonLakeFlyer, 2024-02-09:
I vaguely seem to remember fixes in the qmlsink thingy for Windows. Maybe we lost those somewhere along the way?

NiiAAip, 2024-02-24:
I obtained some important information through https://www.qt.io/blog/graphics-in-qt-6.0-qrhi-qt-quick-qt-quick-3d. Qt Quick aims to accelerate rendering using hardware. By default, it will be built on the low-level graphical API that is most suitable for the target platform. For example, on Windows, it will default to Direct3D, while on macOS, it will default to Metal. 

So here, this line of code cannot run properly.
https://github.com/mavlink/gst-plugins-good/blob/5ff0a6af0e1fd29bafc9f6644e728d24349d0ce7/ext/qt6/qt6glitem.cc#L527

In this way, we can simply bypass it by adding `QQuickWindow:: setGraphicsApi (QSGRenderInterface:: GraphicsApi:: OpenGL);` to the main function. I successfully verified it on my MacBook m2. I think Windows can do the same. On Windows, it may also be possible to implement it through `QCoreApplication: setAttribute (Qt:: AA_UseOpenGLES);` or `QCoreApplication: setAttribute (Qt:: AA_UseSoftwareOpenGL);`.

However, I am not sure if doing so is correct. Does it have an impact on performance?

zdanek, 2024-02-24:
@NiiAAip great finding! Thank you.
I will check it with Windows. I don't have an access to Mac. Can you create a PR with a fix?

DonLakeFlyer, 2024-02-27:
I switched Mac builds to OpenGL. I also tested and it works there at least.

TishSerg, 2024-04-25:
+1 for this. I have the same issue: video still doesn't work as of a commit d785993 (master). 

At the same time, builds of Stable branches have video working ok (at least UDP RTP h264/h265, RSTP; MPEG-TS video isn't shown, but can be recorded with button).

To make video work in master build I've tried changing decoder from `decodebin3` to `decodebin` in `GstVideoReceiver.cc` and `QQuickWindow::setGraphicsApi(QSGRendererInterface::OpenGL);` in `main.cc`.
Still no luck :'(

Windows 11 23H2, Qt 6.6.1/5.15.2 MSVC2019 64bit

AlexKlimaj, 2024-07-03:
Latest daily is just spamming gst_element_factory_make when rtsp is enabled. 

[image]

doquangquan93, 2024-08-16:
Hello everyone, has anyone resolved this issue yet?

TishSerg, 2024-08-16:
> Hello everyone, has anyone resolved this issue yet?

See #11421

doquangquan93, 2024-08-17:
> > Hello everyone, has anyone resolved this issue yet?
> 
> See #11421

Thank you very much! Luckily, it worked

ChaoYuChang, 2024-11-14:
> Latest daily is just spamming gst_element_factory_make when rtsp is enabled. [...]

Sorry to interrupt you, but did you solve the problem and how you solved that? I am facing the same error right now.

s5377689, 2025-03-07:
> Sorry to interrupt you, but did you solve the problem and how you solved that? I am facing the same error right now.

Have you solved this error?
Tried the following code but it didn't work
[image]

[D] at D:\qgroundcontrol\src\QGCApplication.cc:162 - "Settings location "C:/Users/user/AppData/Roaming/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true"
[D] at D:\qgroundcontrol\src\Utilities\QGCLoggingCategory.cc:117 - "Filter rules "*Log.debug=false\nqgc.*.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false""
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:181 - "Using Qt multimedia with FFmpeg version 7.1 LGPL version 2.1 or later"
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:184 - "Available HW decoding frameworks:"
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:186 - "     d3d11va"
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:186 - "     dxva2"
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:186 - "     d3d12va"
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:188 - "Available HW encoding frameworks:"
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:190 - "     d3d11va"
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:190 - "     dxva2"
[I] at C:\Users\qt\work\qt\qtmultimedia\src\plugins\multimedia\ffmpeg\qffmpegmediaintegration.cpp:190 - "     d3d12va"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:113 - "gst_element_factory_make('tee') failed"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:219 - "Failed"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:113 - "gst_element_factory_make('tee') failed"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:219 - "Failed"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:113 - "gst_element_factory_make('tee') failed"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:219 - "Failed"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:113 - "gst_element_factory_make('tee') failed"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:219 - "Failed"
[!] at D:\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\GstVideoReceiver.cc:113 - "gst_element_factory_make('tee') failed"

DonLakeFlyer, 2025-03-07:
That almost always means that the gstreamer plugins couldn't be found.

s5377689, 2025-03-07:
If I compile QGC stable_v4.4 with QT5.12.2, the RTSP display will work correctly. 
If GStreamer is not installed, I will encounter the following error messages.

C:\Program Files\CMake\share\cmake-3.31\Modules\FindPackageHandleStandardArgs.cmake:233: error: Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
Reason given by package: The command
  "C:/gstreamer/1.0/msvc_x86_64/bin/pkg-config.exe" --dont-define-prefix --define-variable=prefix=C:/gstreamer/1.0/msvc_x86_64 --define-variable=libdir=C:/gstreamer/1.0/msvc_x86_64/lib --define-variable=includedir=C:/gstreamer/1.0/msvc_x86_64/include --version
failed with output:

closed 2025-05-03 by HTRamsey