mavlink/qgroundcontrol #11037: Building Error

arunba1, 2024-02-08:
Hi Everyone. while building, I am facing an error, please help me out.

I had cloned the Qgrondcontrol repo using the following command "git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git"

then proceeded with "git submodule update --recursive"
After that, while building with Qt, this error is coming

[image]

[image]

Along with this please help me to add a new button at the toolbar section where can I find that, also please help on how to enlarge a toolbar section with a click and how to revert it back with a double click

atesahmet0, 2024-02-12:
Seems like a common issue #10675 and #10688

Try using:
`git submodule update --init --recursive`

arunba1, 2024-02-12:
> I had Tried using: `git submodule update --init --recursive`
Still it's giving the same error result. Actually I'm using Qt creator 6.6.1 and cloned this repo using git clone and then ran git module update command still result is same.  I'm using Windows11 as OS and compiler as MSVS 2019 64 bit.  please help me out.

Do we need to clone any specific stable version of repo or normal cloning is enough (git clone 'https ://github.com/mavlink/qgroundcontrol.git'

arunba1, 2024-02-15:
Hi All!!
Please send Discord Server link or Forum link .

DonLakeFlyer, 2024-02-22:
QGC: https://discord.com/channels/1022170275984457759/1022185820683255908

Delete your build directory and build again clean.

closed 2024-02-22 by DonLakeFlyer

uby56, 2024-08-29:
hi i am having the same issue can anyone tell me how to solve this problem