mavlink/qgroundcontrol #11039: Pixhawk6X-RT firmware update process fails

dk7xe, 2024-02-09:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
When doing FW update for Pixhawk 6X-RT via QGC the PX4 developer build (master) nightly napshot should be fetched.

## Current Behavior
QGC is reporting "no board specific firmware can be found" in the FW update process

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QGC
2. Go to "Vehicle Setup" / "FirmwareT tab
3. Connect Pixhawk6X-RT via USB
4. FMU is detected as PX4 FMU V6-RT with board ID 35 in QGC (Device USB\VID_3643&PID_001D\0 in Windows device manager)
5. Set "Advanced setting" checkbox on "Firmware Setup" pop up window and select "Developer Build (master)" as FW to be installed.
6. Click "OK" on "Firmware Setup" pop up window
7. QGC reports "Unable to find specific firmware for board type" , "Upgrade cancelled"

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: v4.3.0
- QGC build: stable
- Flight Controller: Pixhawk6X-RT
- Autopilot (with version): PX4 developer build (master)

## Detailed Description
Process has been tested with several versions of FMUM-RT7 / Pixhawk 6X-RT
FMU is detected as Device USB\VID_3643&PID_001D\0

## Log Files and Screenshots
[image]
[image]

dk7xe, 2024-02-09:
https://github.com/mavlink/qgroundcontrol/pull/10846
https://github.com/mavlink/qgroundcontrol/pull/10921

dk7xe, 2024-02-09:
@davids5 @mrpollo

davids5, 2024-02-09:
@dk7xe See https://github.com/mavlink/qgroundcontrol/blame/master/src/VehicleSetup/FirmwareUpgradeController.cc#L65 is is set to 35. Are you using a daily or some old version of QGC?

dk7xe, 2024-02-09:
> @dk7xe See https://github.com/mavlink/qgroundcontrol/blame/master/src/VehicleSetup/FirmwareUpgradeController.cc#L65 it is set to 35. Are you using a daily or some old version of QGC?

QGC Version: v4.3.0
QGC build: stable

That was the first stable with Pixhawk6X-RT support

dk7xe, 2024-03-03:
With latest QGC daily it is working. Please provide a new stable QGC release.

dk7xe, 2024-05-23:
@mrpollo when is the next QGC release planned that adds support for Pixhawk6X-RT?

dk7xe, 2024-07-08:
Solved with QGC release 4.4.0

closed 2024-07-08 by dk7xe