mavlink/qgroundcontrol #11060: Unable to run daily .AppImage  (missing libQt6Quick3D.so.6)

AndKe, 2024-02-15:
## Current Behavior

```
$ ./QGroundControl.AppImage 
/tmp/.mount_QGroun9KlhNB/QGroundControl: error while loading shared libraries: libQt6Quick3D.so.6: cannot open shared object file: No such file or directory

```

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. download daily 
2.run

## System Information
Ubuntu 23.10

## Detailed Description
I've also read https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/index.html
The "Quick 3D" is not selectable for 6.6.1 as described:

[image]

AndKe, 2024-02-15:
@HTRamsey Oh, I know that .AppImage should have what it need, it is actually the other way around - I tried to build it from git first, then saw the message, and then tried .AppImage 

The reason I wanted to try your build is that when I built current HEAD, I did not get the elevation for a mission to work(the yellow line is just flat). - wished to try the new DEM source..

AndKe, 2024-02-20:
@omid-esrafilian Thank you, I did not expect to find it in archive, and then using "filter" was super-counter intuitive, but it worked fine.
Please add this information to build-instructions.

omid-esrafilian, 2024-02-21:
@AndKe 
Yes, indeed it is misleading. I have included this information in my latest pull request #11048.

DonLakeFlyer, 2024-02-22:
AppImage should be fixed now as far as including the various missing quick3d libs. As are docs for building.

closed 2024-02-22 by DonLakeFlyer

AndKe, 2024-02-22:
Yes, thank you. AppImage is working fine now  (except for the https://github.com/mavlink/qgroundcontrol/issues/11084 issue)