mavlink/qgroundcontrol #11077: QGC Linux Appimage fail with Wayland / FC39 KDE

luc-syera, 2024-02-17:
## Expected Behavior
Normal execution 

## Current Behavior
background process are running, but not display

## Steps to Reproduce:
1. Launch QGC appimage 

## System Information

- Fedora FC39 + KDE / Dell precision 7720 i7
- QGC Version: all stable tested versions above, and including 4.2.1 are failling , previous ones are OK
- QGC build: stable versions
- Flight Controller: NA
- Autopilot (with version): NA

## Detailed Description
When launching the appimage ( version >= 4.2.1 stable ) nothing append ( no display )
some background process are running
when launching from a terminal, the following errors concerning Wayland appears : 

```
qt.qpa.wayland: No shell integration named "xdg-shell" found
qt.qpa.wayland: No shell integration named "xdg-shell-v6" found
qt.qpa.wayland: No shell integration named "wl-shell" found
qt.qpa.wayland: No shell integration named "ivi-shell" found
qt.qpa.wayland: Loading shell integration failed.
qt.qpa.wayland: Attempted to load the following shells ("xdg-shell", "xdg-shell-v6", "wl-shell", "ivi-shell")
```

## Log Files and Screenshots

rgreenaus, 2024-11-20:
This problem is still happening and will become more prevalent as ubuntu and other distros move towards wayland as default.

HTRamsey, 2024-11-30:
That's because AppImage doesn't support wayland

julianoes, 2025-03-27:
It works with XWayland:

```
QT_QPA_PLATFORM=xcb ./QGroundControl.AppImage
```

closed 2025-05-03 by HTRamsey