mavlink/qgroundcontrol #11084: QGC Daily for linux - TLS error / unable to fetch elevation/Firmware etc

AndKe, 2024-02-20:
## Current Behavior
The great new elevation is not working due to TLS errors (same for firmware update)

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. get daily .AppImage
2. try to get firmware/elevation
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 23.10
- QGC Version: daily
- 
[image]

## Log Files and Screenshots
Attaching screenshot due to "save" not working.  https://github.com/mavlink/qgroundcontrol/issues/11083

[image]

AndKe, 2024-02-20:
Addtional information;  Same issue exists if I build from source:

gcc version 13.2.0 (Ubuntu 13.2.0-4ubuntu3
QMake version 3.1
Using Qt version 6.6.1 in /home/andre/Qt/6.6.1/gcc_64/lib

HTRamsey, 2024-02-22:
What openssl do you have when building from source

AndKe, 2024-02-22:
@HTRamsey 
$ openssl version 
OpenSSL 3.0.10 1 Aug 2023 (Library: OpenSSL 3.0.10 1 Aug 2023)

HTRamsey, 2024-03-01:
This should be fixed after #11167

DonLakeFlyer, 2024-04-05:
@AndKe Can you verify?

AndKe, 2024-04-06:
@DonLakeFlyer  - yes, the issue was resolved 1 March, works great, and current .AppImage contains the fix too.
Thank you.