mavlink/qgroundcontrol #11088: QGC Stable V4.3 custom MAVLINK message cannot send msg to mocklink

HETONGAPP, 2024-02-20:
Hi there,

1. I switch the V4.3, then I custom the mavlink msg using the mavlink tool, copy the file to the  GQC mavlink lib
2. set vehicle and mocklink file 

previous V4.2 worked well.

Best

DonLakeFlyer, 2024-02-21:
Not sure what you are doing. If you create a custom version of mavlink headers with your own messages. You need to recompile QGC to get it to show up. There is no mention of doing that. When you say cannot send message to mock link. Send message from where?

DonLakeFlyer, 2024-02-23:
If you can compile you new code to send the new message then that shows you have at least added the new mavlink headers correctly. Other than that you are going to need to debug it yourself. We can't help with your own changes to QGC not working. Why it worked before and not now I don't know.

DonLakeFlyer, 2024-02-23:
I would suggest tracing through vehicle to MavinkProtocol to Mocklink to figure it out.

closed 2024-02-23 by DonLakeFlyer