mavlink/qgroundcontrol #11104: Build QGroundControl in container.

liyunfeng159, 2024-02-23:
I'm following the Developer's Guide of QGroundControl.

For the part of build using Containers.
I was able to make the docker image by the docker file in /deploy/docker/Dockerfile-build-linux.

but when I tried to build the android apk file with it. the error shows like:
[image]

I know QGC has been updated to QT 6.6, but the docker file seems like still install the QT 5.15.2.
Is that a bug? or building with docker has been deprecated?
how should I fix that?
Thank you.

futurewebsites, 2024-02-23:
It will need updating in the dockerfile and the linux_qt_install shell script I would think

DonLakeFlyer, 2024-02-23:
Dockerfile needs updating. Please submit a pull to fix it.

commit 6694c97a (2024-02-25): Update for Qt 6.6.1, resolves #11104
closed 2024-02-25 by DonLakeFlyer