mavlink/qgroundcontrol #11118: Determine ArduPilot default hidden flight modes

DonLakeFlyer, 2024-02-25:
The daily build version of QGC has the ability to hide/show flight from the default flight mode dropdown:
[image]

The set of flight mode which are hidden are stored in a setting. That setting for hidden flight modes is currently default to an empty list. I would like help determining what the default for this should be. Or in other words which flight modes are infrequently used and because of that are just adding noise/confusion for most ArduPilot users.

For example in copter the list is massive when the user goes to change flight modes from the toolbar:
[image]

That makes it next to unusable. What should the default much shorter (I hope!) list be? Comment please...

Davidsastresas, 2024-02-25:
I assigned this one to myself. Actually it's been in my to do list for weeks, but I want to first finish that gimbal PR, as it is taking forever.

Ardupilot implemented a parameter that manages precisely this at the AP level, rejecting the mode change if it is configured to do so using this parameter. So I think it would be great to read that parameter, in the newer firmwares that support it, and update automatically the flight mode list based on it. If we want to make it super neat, we could try to detect discordances between the QGC settings and AP parameter and warn the user and hopefully tidy all up for them.

I Think over the next month I will be able to go through this. In the meantime, If you want to leave by default just some flight modes, maybe I would say they should be:

- STABILIZE
- ALT HOLD
- LOITER
- POS HOLD
- GUIDED
- AUTO
- RTL
- LAND

I think the above are a good starting point. For Plane these would be good:

- FBW A
- LOITER
- GUIDED
- RTL
- TAKEOFF

and for VTOL, we should add these to the ones for plane:

- Q LOITER
- Q RTL
- Q HOVER

DonLakeFlyer, 2024-02-26:
Thanks, that is helpful. It also reminds me that ArduPilot flight modes differ quite a bit with respect to vehicle type. I think I'm going to need to change the code to store hidden flight modes by vehicle type before I can make this all work well.

Davidsastresas, 2024-03-04:
I think for Ardupilot we could do something based on that parameter functionality instead, so effectively relying on the vehicle, instead of QGC settings, to show those flight modes. I think we could contain this in APMFirmwarePlugin.

I assigned this to me, I think this month I could take a look at it. I will keep you updated about it, thanks!

DonLakeFlyer, 2024-04-05:
This feature needs to be changed to be vehicle class specific.

DonLakeFlyer, 2024-04-06:
@Davidsastresas I've updated to your feedback

closed 2024-04-06 by DonLakeFlyer

Davidsastresas, 2024-09-17:
Sorry @DonLakeFlyer, I missed this thread all this time and I just came back to it now. It looks great, thank you very much.