mavlink/qgroundcontrol #11127: Cmake builds no longer working

DonLakeFlyer, 2024-02-26:
Output:
```
[main] Building folder: qgroundcontrol 
[main] Configuring project: qgroundcontrol 
[proc] Executing command: /opt/homebrew/bin/cmake --no-warn-unused-cli -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_INSTALL_PREFIX:STRING=/Users/don/Qt/6.6.1/macos -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/clang -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/clang++ -S/Users/don/repos/qgroundcontrol -B/Users/don/repos/qgroundcontrol/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- The C compiler identification is AppleClang 15.0.0.15000100
[cmake] -- The CXX compiler identification is AppleClang 15.0.0.15000100
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: /usr/bin/clang - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: /usr/bin/clang++ - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Daily Build
[cmake] -- lib hints /Users/don/Qt/6.6.1/clang_64;/6.6.1/clang_64;C:/Qt
[cmake] -- Build Type: Debug
[cmake] -- Qt version: 6.6.1
[cmake] -- Qt spec: clang_64
[cmake] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[cmake] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
[cmake] -- Found Threads: TRUE  
[cmake] -- Performing Test HAVE_STDATOMIC
[cmake] -- Performing Test HAVE_STDATOMIC - Success
[cmake] -- Found WrapAtomic: TRUE  
[cmake] -- Found OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.2.sdk/System/Library/Frameworks/OpenGL.framework   
[cmake] -- Found WrapOpenGL: TRUE  
[cmake] -- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-145)") 
[cmake] -- Submodule update
[cmake] -- QGroundControl version: v4.3.0-272-g871875b79
[cmake] -- Enabling video streaming support
[cmake] -- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2") 
[cmake] -- Checking for modules 'gstreamer-1.0>=1.20;gstreamer-video-1.0>=1.20;gstreamer-gl-1.0>=1.20'
[cmake] --   No package 'gstreamer-1.0' found
[cmake] --   No package 'gstreamer-video-1.0' found
[cmake] --   No package 'gstreamer-gl-1.0' found
[cmake] -- GStreamer libs: 
[cmake] -- GStreamer include dirs: 
[cmake] -- GStreamer link dirs: 
[cmake] -- GStreamer cflags: 
[cmake] -- GStreamer ldflags: 
[cmake] -- GStreamer libs: 
[cmake] -- gst found 
[cmake] Building tests
[cmake] -- CMake version = 3.28.3
[cmake] -- CMAKE_SYSTEM_NAME = Darwin
[cmake] -- CMAKE_INSTALL_LIBDIR lib
[cmake] CMake Deprecation Warning at libs/shapelib/CMakeLists.txt:181 (cmake_policy):
[cmake]   The OLD behavior for policy CMP0026 will be removed from a future version
[cmake]   of CMake.
[cmake] 
[cmake]   The cmake-policies(7) manual explains that the OLD behaviors of all
[cmake]   policies are deprecated and that a policy should be set to OLD only under
[cmake]   specific short-term circumstances.  Projects should be ported to the NEW
[cmake]   behavior and not rely on setting a policy to OLD.
[cmake] 
[cmake] 
[cmake] -- WARNING: EG_DATA:PATH not set to point to downloaded eg_data directory so the eg_data part of testing will be ignored.
[cmake] sed: script.sed: No such file or directory
[cmake] sed: script.sed: No such file or directory
[cmake] sed: script.sed: No such file or directory
[cmake] CMake Deprecation Warning at libs/qmdnsengine/CMakeLists.txt:1 (cmake_minimum_required):
[cmake]   Compatibility with CMake < 3.5 will be removed from a future version of
[cmake]   CMake.
[cmake] 
[cmake]   Update the VERSION argument <min> value or use a ...<max> suffix to tell
[cmake]   CMake that the project does not need compatibility with older versions.
[cmake] 
[cmake] 
[cmake] -- QMdnsEngine: Found Qt Version: 6.6.1
[cmake] -- QMdnsEngine: Qt version used: 6.6.1
[cmake] -- QMdnsEngine: Enable C++17 for Qt 6.6 and newer
[cmake] -- UTMSP: Dummy is Initialized
[cmake] -- Configuring done (15.7s)
[cmake] CMake Error: The inter-target dependency graph contains the following strongly connected component (cycle):
[cmake]   "libevents_parser" of type STATIC_LIBRARY
[cmake]     depends on "comm" (weak)
[cmake]     depends on "qgc" (weak)
[cmake]     depends on "ADSB" (weak)
[cmake]     depends on "AnalyzeView" (weak)
[cmake]     depends on "api" (weak)
[cmake]     depends on "Audio" (weak)
[cmake]     depends on "AutoPilotPlugins" (weak)
[cmake]     depends on "Camera" (weak)
[cmake]     depends on "compression" (weak)
[cmake]     depends on "FactSystem" (weak)
[cmake]     depends on "FirmwarePlugin" (weak)
[cmake]     depends on "FlightMap" (weak)
[cmake]     depends on "FollowMe" (weak)
[cmake]     depends on "Geo" (weak)
[cmake]     depends on "gps" (weak)
[cmake]     depends on "Joystick" (weak)
[cmake]     depends on "MissionManager" (weak)
[cmake]     depends on "PositionManager" (weak)
[cmake]     depends on "QmlControls" (weak)
[cmake]     depends on "QtLocationPlugin" (weak)
[cmake]     depends on "Settings" (weak)
[cmake]     depends on "Terrain" (weak)
[cmake]     depends on "uas" (weak)
[cmake]     depends on "UTMSP" (weak)
[cmake]     depends on "Vehicle" (weak)
[cmake]     depends on "VehicleSetup" (weak)
[cmake]     depends on "VideoManager" (weak)
[cmake]     depends on "Viewer3D" (weak)
[cmake]     depends on "AirLink" (weak)
[cmake]     depends on "qgctest" (weak)
[cmake]     depends on "AnalyzeViewTest" (weak)
[cmake]     depends on "qgcunittest" (weak)
[cmake]     depends on "AudioTest" (weak)
[cmake]     depends on "FactSystemTest" (weak)
[cmake]     depends on "GeoTest" (weak)
[cmake]     depends on "MissionManagerTest" (weak)
[cmake]     depends on "QmlControlsTest" (weak)
[cmake]     depends on "VehicleTest" (weak)
[cmake]     depends on "libevents_parser_autogen_timestamp_deps" (strong)
[cmake]     depends on "libevents_parser_autogen" (strong)
[cmake]   "libevents_health_and_arming_checks" of type STATIC_LIBRARY
[cmake]     depends on "comm" (weak)
[cmake]     depends on "qgc" (weak)
[cmake]     depends on "ADSB" (weak)
[cmake]     depends on "AnalyzeView" (weak)
[cmake]     depends on "api" (weak)
...
```

Do you think you could move at least one OS over to cmake in CI? That way we have coverage on keeping cmake working. I'd rather see that before any more large scale cmake structure changes.

DonLakeFlyer, 2024-02-26:
If you do one, I can move it over to OSX which is my main platform. If you don't have that.

DonLakeFlyer, 2024-02-26:
> Do you think you could move at least one OS over to cmake in CI?

Let me know if you don't have time and I'll take a stab at it. Maybe just a build step without a deploy for now as an extra action.

HTRamsey, 2024-02-26:
Oh, qt_add_library has different defaults than add_library for Shared vs Static. I have to manually specify for each. Yeah I'll get a linux and android cmake set up in CI today.

s-lisovenko, 2024-02-26:
Fixed in this PR
https://github.com/mavlink/qgroundcontrol/pull/11128

DonLakeFlyer, 2024-02-26:
Thanks to both of you

HTRamsey, 2024-02-26:
That was insanely quick but yeah that was exactly what I was going to do, PR looks good

DonLakeFlyer, 2024-02-26:
I'm old, so hence old-school dev. cmake is all new to me! So can't quite jump in quickly to fix things. I'll come up to speed as some point :)

closed 2024-02-26 by DonLakeFlyer