mavlink/qgroundcontrol #11140: MPEG-TS H.264 stream is not shown

ddatsko, 2024-02-27:
MPEG-TS H.264 stream is not shown in QGC, however the video can be recorded well
----

## Expected Behavior
- When video source is set to `MPEG-TS (h.264)` and `UDP port` is set to 8081, the video should be shown in QGC after running
`gst-launch-1.0 videotestsrc ! video/x-raw, format=NV12, width=1920, height=1080 ! x264enc ! h264parse config-interval=-1 ! mpegtsmux ! udpsink host=127.0.0.1 port=8081`

## Current Behavior
When performing the steps above, the video is not shown in QGC. However, when the stream is running, the "Record" button becomes active and is fully functional. After recording the video, the `.mkv` file with recording contains the valid received video

## Steps to Reproduce:
1. In QGC, set up video source to `MPEG-TS (h.264)` and UDP port to `8081`
2. Run the gstreamer pipeline from above

## System Information
- Operating System: Ubuntu 22.04
- QGC Version:
  -   v4.3.0 release from github
  -  Build from source in current `master`

## Log Files and Screenshots
- Log of QGroundControl launched when the stream is already running: [QGC_log.txt](https://github.com/mavlink/qgroundcontrol/files/14424457/QGC_log.txt)

ddatsko, 2024-02-28:
UPD: I was debugging the issue more deeply, and it looks that the problem is in the `decodebin3` gstreamer plugin. When the decoder is set to `decodebin` instead of `decodebin3` in the code, the video is played and I obtain the following pipeline inside of the QGC with properly selected `avdec_h264` decoder: 
[image]
However, if I keep the `decodebin3`, the pipeline is created only till this stage and looks like `decodebin3` cannot properly handle the stream
[image]
All this also fits well with the observation that video recording works well: there is just no decoding there and the stream goes directly into `matroskamux` and the file

fredowski, 2024-03-05:
I dived into this 100m deep tank of shit of the gstreamer video stuff some time ago: https://github.com/mavlink/qgroundcontrol/pull/10420 

My result is that it is impossible to setup a automagic pipeline that works for all video formats on all platforms. So if you figure out one pipeline setup that works for your particular video on your platform then you have to make sure that this works on all other platforms: linux, MacOS, Windows, Android. Then you have to make sure that you have not broken other video formats.

ericjohnson97, 2024-03-22:
I wanted to +1 this issue. I am seeing the same error in my QGC log on Windows when trying to receive a UDP h264 stream

```
[!] at D:\a\qgroundcontrol\qgroundcontrol\src\VideoReceiver\GstVideoReceiver.cc:1305 - "GStreamer error: Internal data stream error."
```

closed 2026-10-04 by github-actions[bot]