mavlink/qgroundcontrol #11159: Can't add ROI

SebastianViollaz, 2024-02-29:
----

## Expected Behavior
Load a simple mission with ROI

## Current Behavior
Can't add  ROI: GIven alert:
[image]
Al mission items:
[image]
This is the mission:
[image]

## Steps to Reproduce:

1. Add a mission with a roi and a cancel roi item
2. Load the mmission

## System Information

- Operating System: Linux Ubuntu 22.04LTS
- QGC Version: Last stable
- QGC build: daily build
- Flight Controller:Simulated ArduCopeter, and in a PIX4 Ardupilot Copter  (Last version)

SebastianViollaz, 2024-02-29:
Sure, i can test it, are you merging to master?

DonLakeFlyer, 2024-02-29:
> re you merging to master?

I'd rather have you give it a try first

SebastianViollaz, 2024-03-01:
ok, i will test it, just let me known when it'd done

SebastianViollaz, 2024-03-01:
I saw your PR and i downloaded the branch. This error is now given:
[image]

THis is the complete mission:
[image]

DonLakeFlyer, 2024-03-03:
@SebastianViollaz What firmware type/version?

SebastianViollaz, 2024-03-04:
I sorry, for the delay, i solve in the ardupilot
 STL updating to the master brach version. I will check de version in the uav but i think it's the last stable

SebastianViollaz, 2024-03-04:
Now
 have the 4.4.4 ArduCopter firmware and now its working

PR #11161 (2024-03-04): DO_SET_ROI_LOCATION is now supported
This fixes a bug with ArduPilot complaining about NANs in cancel roi. Fix #11159
closed 2024-03-04 by DonLakeFlyer