mavlink/qgroundcontrol #11169: Qt6.6.1 Android compile error on windows

anhhien2k, 2024-03-01:
I tried successfully with windows version with qt6.6 but when I use android it got an error:
"make: *** No rule to make target 'C:\source\QGC\build-qgroundcontrol-Android_Qt_6_6_2_Clang_arm64_v8a-Debug\ANDROID_PACKAGE_SOURCE_DIR\AndroidManifest.xml', needed by 'libQGroundControl_arm64-v8a.so'.  Stop."

This is a kit information:
[image]

And this is the error:
[image]

Please help me solve this problem!

HTRamsey, 2024-03-01:
kinda weird that it looks like it was running the "which ccache" command on windows. Remove the following code from QGCCommon.pri and try again.
```
# Enable ccache where we can
linux|macx|ios {
    system(which ccache) {
        message("Found ccache, enabling")
        !ios {
            QMAKE_CXX = ccache $$QMAKE_CXX
            QMAKE_CC  = ccache $$QMAKE_CC
        } else {
            QMAKE_CXX = $$PWD/tools/iosccachecc.sh
            QMAKE_CC  = $$PWD/tools/iosccachecxx.sh
        }
    }
}
```

anhhien2k, 2024-03-02:
@HTRamsey I tried your suggestion but the error still occurs. I am building for android and i think what you said is not the problem here.

andrefreitas97, 2024-03-04:
I had the same problem. Edit the AndroidManifest.xml file as you want (name of the app... etc) and then copy all the files in qgroundcontrol
/android folder to the C:\source\QGC\build-qgroundcontrol-Android_Qt_6_6_2_Clang_arm64_v8a-Debug\ANDROID_PACKAGE_SOURCE_DIR folder and it should work

anhhien2k, 2024-03-05:
@andrefreitas97 This is the error:
make: *** No rule to make target 'C:\source\QGC\build-qgroundcontrol-Android_Qt_6_6_1_Clang_arm64_v8a-Debug\ANDROID_PACKAGE_SOURCE_DIR\AndroidManifest.xml', needed by 'libQGroundControl_arm64-v8a.so'.  Stop.
The process "C:\Install\Android\Sdk\ndk\25.1.8937393\prebuilt\windows-x86_64\bin\make.exe" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Android Qt 6.6.1 Clang arm64-v8a)
When executing step "Make"

There are 2 folder in build-qgroundcontrol-Android_Qt_6_6_1_Clang_arm64_v8a-Debug: 
- .qm
- .qtc_clangd
 
I don't know the ANDROID_PACKAGE_SOURCE_DIR  folder?

anhhien2k, 2024-03-05:
@andrefreitas97 I followed your instructions and it worked. However, it shows another error related to UnitTest:

```
C:\source\QGC\qgroundcontrol\src\main.cc:386:28: error: use of undeclared identifier 'UnitTest'
            int failures = UnitTest::run(unitTestOptions);
```

I tried changing it to OFF in the `CMakeList.txt` file:
```
cmake_dependent_option(QGC_BUILD_TESTING "Enable testing" OFF "CMAKE_BUILD_TYPE STREQUAL Debug" OFF)
```
but this error still occurs.

pls help me to solve this problem!

HTRamsey, 2024-03-05:
This is fixed in #11154 (unit tests originally weren't meant for mobile). Just build via qmake until cmake is ready.

anhhien2k, 2024-03-05:
@HTRamsey  I followed your instructions and it shows another error:
```
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\ScreenTools\ScreenTools.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\ClickableColor.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\DeadMouseArea.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\ParameterEditor.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\JoystickThumbPad.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\GPSIndicatorPage.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\MissionItemStatus.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\FactSliderPanel.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\OfflineMapButton.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\TerrainStatus.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\QGCMenuSeparator.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\QGCMapCircleVisuals.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\SubMenuButton.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\QGCColoredImage.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\QGCMenu.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\MissionItemIndexLabel.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\MissionItemMapVisual.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\QGCLabel.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\ToolIndicatorPage.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\LabelledComboBox.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\FlightModeDropdown.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\QGCTextField.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\MainWindowSavedState.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\PipView.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\AppMessages.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\LogReplayStatusBar.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\FlightModeMenuIndicator.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\ExclusiveGroupItem.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\FactSlider.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\InstrumentValueLabel.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\OfflineMapEditor.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\QGCSlider.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\QGCRoundButton.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\FileButton.qml:The system cannot find the file specified.
Cannot open input file C:\source\qgroundcontrol\src\QmlControls\QGroundControl\Controls\SettingsGroupLayout.qml:The system cannot find the file specified.
[... cut ...]
Warning: QML import could not be resolved in any of the import paths: Viewer3D.Models3D.Drones
Failed to copy C:/source/build-qgroundcontrol-Android_Qt_6_6_1_Clang_arm64_v8a-Debug/android-build/AndroidManifest.xml to C:/source/build-qgroundcontrol-Android_Qt_6_6_1_Clang_arm64_v8a-Debug/android-build/AndroidManifest.xml.
12:27:43: The process "C:\Qt\6.6.1\mingw_64\bin\androiddeployqt.exe" exited with code 11.
Error while building/deploying project qgroundcontrol (kit: Android Qt 6.6.1 Clang arm64-v8a)
```
pls help me to solve this problem!

VijaySue, 2024-03-06:
> This is fixed in #11154 (unit tests originally weren't meant for mobile). Just build via qmake until cmake is ready.

I get the same error with qmake, and still can't build in cmake. Is this issue not fixed?

[image]

anhhien2k, 2024-03-06:
@VijaySue With this error, you follow some steps:
- create ANDROID_PACKAGE_SOURCE_DIR folder in E:\QtProject\build-qgroundcontrol-Android_Qt_6_6_2_Clang_arm64_v8a-Debug.
- After, you copy all the files in qgroundcontrol/android folder to the E:\QtProject\build-qgroundcontrol-Android_Qt_6_6_2_Clang_arm64_v8a-Debug\ANDROID_PACKAGE_SOURCE_DIR folder and build again.

Zameer-fatima, 2024-03-13:
Heyy!!
I am facing the same errors as you have. How to solve this. can you pls help me

VijaySue, 2024-03-13:
With this error, you follow some steps:

create ANDROID_PACKAGE_SOURCE_DIR folder in E:\QtProject\build-qgroundcontrol-Android_Qt_6_6_2_Clang_arm64_v8a-Debug.
After, you copy all the files in qgroundcontrol/android folder to the E:\QtProject\build-qgroundcontrol-Android_Qt_6_6_2_Clang_arm64_v8a-Debug\ANDROID_PACKAGE_SOURCE_DIR folder and build again.

---- Replied Message ----
| From | Zameer ***@***.***> |
| Date | 3/13/2024 14:06 |
| To | ***@***.***> |
| Cc | ***@***.***>,
***@***.***> |
| Subject | Re: [mavlink/qgroundcontrol] Qt6.6.1 Android compile error on windows (Issue #11169) |

Heyy!!
I am facing the same errors as you have. How to solve this. can you pls help me

Zameer-fatima, 2024-03-15:
@VijaySue 
Hii !!
Thanks for your response & suggestion, It worked and that errors were gone But it is still showing the error I have mentioned below.
How to solve this one??, pls guide...

But at last the Build Successfull is displaying and it has errors with emulator.

[image]

[image]

VijaySue, 2024-03-15:
You should use Android clang compiler instead of mingw.

---- Replied Message ----
| From | Zameer ***@***.***> |
| Date | 03/15/2024 21:19 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [mavlink/qgroundcontrol] Qt6.6.1 Android compile error on windows (Issue #11169) |

@VijaySue
Hii !!
Thanks for your response & suggestion, It worked and that errors were gone But it is still showing the error I have mentioned below.
How to solve this one??, pls guide...

But at last the Build Successfull is displaying and it has errors with emulator.

image.png (view on web)

image.png (view on web)

tranhien1612, 2024-03-15:
@VijaySue Did you build successfully?

VijaySue, 2024-03-15:
Can your emulater start directly? Other than start by run project.

---- Replied Message ----
| From | Zameer ***@***.***> |
| Date | 03/15/2024 21:19 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [mavlink/qgroundcontrol] Qt6.6.1 Android compile error on windows (Issue #11169) |

@VijaySue
Hii !!
Thanks for your response & suggestion, It worked and that errors were gone But it is still showing the error I have mentioned below.
How to solve this one??, pls guide...

But at last the Build Successfull is displaying and it has errors with emulator.

image.png (view on web)

image.png (view on web)

VijaySue, 2024-03-15:
Yes.

---- Replied Message ----
| From | Trần Bá ***@***.***> |
| Date | 03/16/2024 00:26 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [mavlink/qgroundcontrol] Qt6.6.1 Android compile error on windows (Issue #11169) |

@VijaySue Did you build successfully?

ZHCLN, 2024-03-18:
> I tried successfully with windows version with qt6.6 but when I use android it got an error: "make: *** No rule to make target 'C:\source\QGC\build-qgroundcontrol-Android_Qt_6_6_2_Clang_arm64_v8a-Debug\ANDROID_PACKAGE_SOURCE_DIR\AndroidManifest.xml', needed by 'libQGroundControl_arm64-v8a.so'. Stop."
> 
> This is a kit information: ![kit](https://private-user-images.githubusercontent.com/69721832/309237810-c5e63f3e-dbd4-46b9-a309-8790baf880c0.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3MzA0NTIsIm5iZiI6MTcxMDczMDE1MiwicGF0aCI6Ii82OTcyMTgzMi8zMDkyMzc4MTAtYzVlNjNmM2UtZGJkNC00NmI5LWEzMDktODc5MGJhZjg4MGMwLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyNDkxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNhODgxMmFmNTZkZDE1OTM5MDA1MTQ3Mjk4YThkNGQ5ZDcxMzNjNDQwMzhhOTY0MDdiYzlkOTgyMWRiYTlhZDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4LFdjOGhnZA9nkcxYnIXicfOZP1fqQeyEAn04Dls6iM)
> 
> And this is the error: ![error](https://private-user-images.githubusercontent.com/69721832/309237939-fde0a031-f648-4c50-92c3-244307090a6b.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3MzA0NTIsIm5iZiI6MTcxMDczMDE1MiwicGF0aCI6Ii82OTcyMTgzMi8zMDkyMzc5MzktZmRlMGEwMzEtZjY0OC00YzUwLTkyYzMtMjQ0MzA3MDkwYTZiLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAyNDkxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzYTI5MGNhMjE4NzcxYWJmOGI2NjJhNWQyOWYwNDJjNTBhZWVlNDQyNmI1MDBkMjUxMjZjNmI2YjIzMjVjOTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7THFB6zM41WsRJbjR67jd7XpQYf4j2CrB2MK0PKhnT4)
> 
> Please help me solve this problem!

I also encountered the same problem, have you solved it now?

tranhien1612, 2024-03-18:
> @VijaySue With this error, you follow some steps:
>
> * create ANDROID_PACKAGE_SOURCE_DIR folder in E:\QtProject\build-qgroundcontrol-Android_Qt_6_6_2_Clang_arm64_v8a-Debug.
> * After, you copy all the [...]

@ZHCLN pls try it

ZHCLN, 2024-03-31:
> @ZHCLN pls try it

I tried your solution, but I still get other errors.
[image]

[image]

Zameer-fatima, 2024-03-31:
> Can your emulater start directly? Other than start by run project.
> […](#)

I have used physical device. The apk worked in my device. Thank You

Zameer-fatima, 2024-03-31:
@VijaySue
Hii !
Thank you for your suggestions, but now
I aim to develop QGroundControl (QGC) for Android using QT5.15.1, specifically targeting stable version 4.3.0. However, I'm encountering an issue where the arm64_v8a kit is not available in QT5, leading to difficulties in the building process. Could you please advise on how to successfully build it using QT5?

BreederBai, 2024-04-19:
@ZHCLN I also encountered the same error as you. Have you solved this error?
[image]

closed 2024-11-30 by HTRamsey