mavlink/qgroundcontrol #11175: Segfaults while fetching altitude.

AndKe, 2024-03-02:
## Current Behavior
(gdb log further below)

```
andre@Loke:~/prog/qgroundcontrol/build/staging$ ./QGroundControl 
Settings location "/home/andre/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United Kingdom) ; Name "en_GB" ; Preffered (used in maps):  "en-GB"
LocalizationLog: Qt lib localization for "en_GB" is not present
LocalizationLog: Error loading source localization for "en_GB"
LocalizationLog: Error loading json localization for "en_GB"
qt.multimedia.ffmpeg.libsymbolsresolver: Couldn't load VAAPI library
MAVLinkLogManagerLog: MAVLink logs directory: "/home/andre/Documents/QGroundControl Daily/Logs"
Map Cache in: "/home/andre/.cache/QGCMapCache300" / "qgcMapCache.db"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:308:21: QML Rectangle: Binding loop detected for property "height"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:308:21: QML Rectangle: Binding loop detected for property "height"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:307:21: QML Rectangle: Binding loop detected for property "width"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:308:21: QML Rectangle: Binding loop detected for property "height"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:307:21: QML Rectangle: Binding loop detected for property "width"
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error QGeoPositionInfoSource::AccessError
"v4.3.0"
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
Found
Delayed split
Update delayed split segment
setCurrentPlanViewSeqNum
Found
Delayed split
Update delayed split segment
Segmentation fault (core dumped)
andre@Loke:~/prog/qgroundcontrol/build/staging$ ./QGroundControl 
Settings location "/home/andre/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United Kingdom) ; Name "en_GB" ; Preffered (used in maps):  "en-GB"
LocalizationLog: Qt lib localization for "en_GB" is not present
LocalizationLog: Error loading source localization for "en_GB"
LocalizationLog: Error loading json localization for "en_GB"
qt.multimedia.ffmpeg.libsymbolsresolver: Couldn't load VAAPI library
MAVLinkLogManagerLog: MAVLink logs directory: "/home/andre/Documents/QGroundControl Daily/Logs"
Map Cache in: "/home/andre/.cache/QGCMapCache300" / "qgcMapCache.db"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:308:21: QML Rectangle: Binding loop detected for property "height"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:308:21: QML Rectangle: Binding loop detected for property "height"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:307:21: QML Rectangle: Binding loop detected for property "width"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:308:21: QML Rectangle: Binding loop detected for property "height"
qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:307:21: QML Rectangle: Binding loop detected for property "width"
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error QGeoPositionInfoSource::AccessError
"v4.3.0"
"Linear Tech" "LTC DC1613A"
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: -1
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 866419611
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 866419611
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 866419611
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 866419611
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 866419611
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 866419611
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 866419611
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 866419611
QGCMapUrlEngineLog: getProviderTypeFromQtMapId : map id not found: 690512839
Fetch tile error: "Error transferring http://khm3.google.com/kh/v=692&hl=en-GB&x=141&y=59&z=8&s=Ga - server replied: Not Found"
Fetch tile error: "Error transferring http://khm2.google.com/kh/v=692&hl=en-GB&x=140&y=59&z=8&s=Galileo - server replied: Not Found"
Fetch tile error: "Error transferring http://khm0.google.com/kh/v=692&hl=en-GB&x=138&y=59&z=8&s=G - server replied: Not Found"
Fetch tile error: "Error transferring http://khm1.google.com/kh/v=692&hl=en-GB&x=139&y=59&z=8&s=Gali - server replied: Not Found"
Fetch tile error: "Error transferring http://khm2.google.com/kh/v=692&hl=en-GB&x=138&y=60&z=8&s=Ga - server replied: Not Found"
Fetch tile error: "Error transferring http://khm3.google.com/kh/v=692&hl=en-GB&x=139&y=60&z=8&s=Galil - server replied: Not Found"
[... 102 lines cut ...]
[Thread 0x7fffd5dfe6c0 (LWP 9862) exited]
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
[New Thread 0x7fffd5dfe6c0 (LWP 9890)]
setCurrentPlanViewSeqNum
[Thread 0x7fffd5dfe6c0 (LWP 9890) exited]
[New Thread 0x7fffd5dfe6c0 (LWP 9891)]
[New Thread 0x7fffeccfb6c0 (LWP 9892)]
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
Found
Delayed split
Update delayed split segment
setCurrentPlanViewSeqNum
Found
Delayed split
Update delayed split segment
TerrainTileLog: 0x55555d2556b8 Warning: elevation read is below min elevation in tile: 0 < 126

Thread 1 "QGroundControl" received signal SIGSEGV, Segmentation fault.
0x0000555555821d9f in TerrainTile::elevation (coordinate=..., this=0x55555d2556b8) at /home/andre/prog/qgroundcontrol/src/TerrainTile.cc:119
119	    const auto elevation = _data[latIndex][lonIndex];
(gdb) q

```

DonLakeFlyer, 2024-03-04:
@AndKe Can you attach the plan file which causes this?

AndKe, 2024-03-04:
@DonLakeFlyer by "plan file" - do you mean flightplan/mission file?
(at this point I loaded some old missions, and everything worked perfectly.)
- Then I zoomed out, and manually added some waypoints like before:
It seems to me that wp distance on 10km is ok, 20km causes crash:
Please try to load this:
[mission.txt](https://github.com/mavlink/qgroundcontrol/files/14485444/mission.txt)

DonLakeFlyer, 2024-03-04:
This logging shows you tile data has some crap in it: TerrainTileLog: 0x55dc16203738 Warning: elevation read is below min elevation in tile: 0 < 209

AndKe, 2024-03-04:
Well, I reproduced it using google-terrain  and bing-sat. right now.
Simple way to reproduce:
set a takeoff point, and two WP(any short distance will do)
Start to drag the second WP, keep an eye on distance, once you pass about 20km, ..crash.
And yes - oen need to handle bad elevation data.
FWIW: most places I tested now are totally random, and unless QGC is precaching some huge areas, there is no reason to believe I "had" those corrupted data before drawing the mission.

DonLakeFlyer, 2024-03-04:
Doesn't repro for me. Most likely because as I said above, you somehow got a bad tile in the cache. It's gonna stay in there now till you clear the DB. How it go there I'm not quite sure. I added a work around to prevent the crash, but you are most likely going to need to clear your map cache. Newer builds will warn on the console instead of crashing on a bad reference. And you should see that you are missing terrain data if you manage to get a bad tile in again.

Other option is zip up your map db and put it somewhere I can download it. Then I can look in there.

AndKe, 2024-03-05:
I guess you are right :)
I built the current HEAD
then did: 
rm -r ~/.cache/QGroundControl.org/
rm -r ~/.cache/QGCMapCache300/

and I cannot reproduce the issue anymore, at least not at this computer.

DonLakeFlyer, 2024-03-05:
Which is kinda disappointing. The question still remains as to how it got screwed up in the first place. Something lurking there still. At least I've hopefully cauterized the crash. I'm going to close this for now until there is some sort of repro.

closed 2024-03-05 by DonLakeFlyer