mavlink/qgroundcontrol #11185: FlightMap zooming issue in Linux

omid-esrafilian, 2024-03-03:
@DonLakeFlyer 
I realized that zooming the map using the Mouse middle button (wheel) does not work in Linux, either in Fly View or Plan View. However, if the acceptedDevices of the WheelHandler in the FlightMap.qml file is changed as follows, the issue can be resolved.

acceptedDevices: PointerDevice.Mouse | PointerDevice.TouchPad

DonLakeFlyer, 2024-03-06:
I don't know. That wheel handler code was taken right out of the Qt Docs for the Map qml item. I assume it correctly works around the bug they describe. You can see it here: https://doc.qt.io/qt-6/qml-qtlocation-map.html. Could. you take a look at that as well as the details on the bug it references and see what you think?

omid-esrafilian, 2024-03-07:
@DonLakeFlyer 
I have reviewed the bug reports, but they are dated from a couple of years ago, so It's unclear whether the issue still exists.
I have also checked the "OrbitCameraController" qtquick3d module provided by Qt for controlling the camera using a mouse or touchpad. It seems that the "WheelHandler" is configured similarly to what I mentioned above, and no reports of any bug.
However, to be on the safe side, I have checked the "Qt.platform.pluginName" under Linux when using the mouse (at least in my case) and it appears to be "xcb". I added this as another exception in the code and it works as expected. Here is the updated code:

acceptedDevices:    Qt.platform.pluginName === "cocoa" || Qt.platform.pluginName === "wayland" || Qt.platform.pluginName === "xcb"? PointerDevice.Mouse | PointerDevice.TouchPad : PointerDevice.Mouse

So, what do you think?

closed 2024-11-30 by HTRamsey