mavlink/qgroundcontrol #11190: ArduPilot Survey Missions not loading in properly

ben-draper-ascent, 2024-03-04:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Survey Missions on Ardupilot vehicle that are preloaded reload into Plan View

## Current Behavior
Survey Missions that are preloaded on an Ardupilot vehicle are registering as unloaded missions in Plan View

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Make survey and upload
2. Close QGC and reopen (without closing SITL)
3. Open QGC and open Plan View (Upload Required will be flashing)

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04
- QGC Version: Master
- QGC build: daily
- Flight Controller: SITL
- Autopilot (with version): ArduPilot 4.3.7

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots

DonLakeFlyer, 2024-03-04:
Dirty bit is getting set.

DonLakeFlyer, 2024-04-05:
This was fixed by the other Pan fixes I made a while back

closed 2024-04-05 by DonLakeFlyer