mavlink/qgroundcontrol #11193: Settings Menu Scrolling Bug

ben-draper-ascent, 2024-03-05:
@DonLakeFlyer 

Scrolling seems to be broken on mobile in the Application Settings. When trying to scroll, its actually dragging the map. You can simulate this on desktop by drag scrolling instead of scroll wheel scrolling

DonLakeFlyer, 2024-03-05:
@ben-draper-ascent Can you try the latest daily. I may have fixed this with a previous pull

ben-draper-ascent, 2024-03-06:
@DonLakeFlyer The scrolling in the settings menu seems to work but I am noticing that depending on where you drag to scroll from theres a chance you can still drag the map around behind it

ben-draper-ascent, 2024-03-06:
@DonLakeFlyer Oh scrolling does not seem to be fixed. At least not on my android device

DonLakeFlyer, 2024-03-06:
I don't quite get what you are doing. I tried scrolling the general settings page and it didn't move the map as far as I can tell. I'm going to need more details. Maybe make a video?

arnaudjuin, 2024-03-07:
Same issue here.

ben-draper-ascent, 2024-03-07:
https://github.com/mavlink/qgroundcontrol/assets/107940501/045f3ad6-b4ed-4728-8e8c-05853ed46d5f

@DonLakeFlyer

ben-draper-ascent, 2024-03-07:
> ScrollBug_lowres.mp4 
> @DonLakeFlyer

This is the Daily as of 2 days ago

DonLakeFlyer, 2024-03-08:
@ben-draper-ascent Can you verify

arnaudjuin, 2024-03-08:
Amazing. Thank you.

PR #11211 (2024-03-08): DragHandler for map was bleeding through to lower controls
Not sure wht. Went back to old way of panning map. Fix #11193