mavlink/qgroundcontrol #11197: Mavlink Inspector: Can not deselect a display item / Checkbox is greyed but still displayed / Graph can not be removed

fredowski, 2024-03-05:
## Expected Behavior

When I click on a checkbox for an item like "Attitude.pitch", then pitch is displayed in the graph vs. time. The checkbox is marked with a tick. When I click again on the now ticked box, the tick is removed and the pitch is removed from the graph display.

## Current Behavior
When I click on the already ticked checkbox, the checkbox is greyed out. The graph is still displayed. It is not possible to remove the item from the graph display. It is only possible to add more graphs.

## System Information
When posting bug reports, include the following information
- Operating System: MacOS 12.7.3 Monterey
- QGC Version: QGC Daily master 1984b00e 2024-03-04
- QGC build: daily
- Flight Controller: SITL
- Autopilot (with version): Arduplane master

## Log Files and Screenshots
[image]
The screenshot shows the correctly marked item "Attitude.pitch" after I clicked once on the checkbox

[image]
The screenshot shows the mavlink inspector state after I clicked a second time on the checkbox. The checkbox is greyed out. The item is not removed from the graph window.

HTRamsey, 2024-03-26:
Whoops I had made a duplicate issue before I saw this. I think this is caused by the `checked` property no longer affecting the `checkable` property. So some of the checks in the `enabled` property need to be merged with the checks in the `checked` property and moved to `checkable`. Then you no longer manually set `checked`. Also, the signal onToggled should probably be used instead of onClicked.

vryabokon1705, 2025-02-25:
Hi. Added PR with workaround: https://github.com/mavlink/qgroundcontrol/pull/12508