mavlink/qgroundcontrol #11201: RTL no showed in the 3D view

SebastianViollaz, 2024-03-06:
Would be nice to add the RTL icon. At the moment the RTL isn't showed and the flight path isn't showed:

[image]

As you can see in the last point doesn't draw a line to the Take off item like in the 2D view.

SebastianViollaz, 2024-03-06:
Yes this is the flight.
[image]

omid-esrafilian, 2024-03-06:
@SebastianViollaz 
Thanks. Can you also please send me the list of your mission that is shown in the Plan View? 
Because it is a bit strange that it does not show the Takeoff point for you.

omid-esrafilian, 2024-03-06:
@SebastianViollaz 
May I ask which operating system (OS) you use?

SebastianViollaz, 2024-03-06:
This are the list items.
[image]
[image]

I'm in linux SO, Ubuntu 22.04.4 LTS

omid-esrafilian, 2024-03-07:
@SebastianViollaz 
I have just created this pull request (#11214). Please let me know if that solves the problem.

closed 2025-05-03 by HTRamsey