mavlink/qgroundcontrol #11202: Add 3D flight editor

SebastianViollaz, 2024-03-06:
Adding a 3d editor would be very useful  if you have a properly designed 3d MAP. If you are in an indoor environment using a 3D edit would be easier, and will help avoiding collisions during the flight planification. Maybe adding  some arrows to move the point like 3D design apps:
[image]

omid-esrafilian, 2024-03-06:
@SebastianViollaz 
Thanks for the suggestion. But do you mean to add an editor to create/edit a map inside the QGC 3D View?

In general, yes that is a good feature, however, it requires extensive work.

SebastianViollaz, 2024-03-06:
Yes, it's a big feature but would be nice maybe someone is interested in developing it

omid-esrafilian, 2024-03-06:
But it is not clear to me why someone needs to edit the map inside the QGC?
The map can be for example created using a professional 3D editor software and then imported to the QGC. Does that make sense?

What in my opinion would make more sense is to generalize the current 3D Viewer to also accept and import other file formats such as STL.

SebastianViollaz, 2024-03-07:
Maybe you misunderstand me, the feature that i propose is to edit that flight path, in the 3d View, not the 3D models of the view.
In this 3D editor you only be able to edit the items positions (Checkpoints,RIOS's, Take Off, RTL etc. 
At the moment you edit in the 2D view, but editing in the 3D view could be nice to determinate the exact position of the points and the exact altitude without having to change to the 3D view to check if the point is in the correct position.

Basically the feature is not a 3D map edit, instead is a 3D flight editor, like the flight editor that QGC already have, but in 3D

SebastianViollaz, 2024-03-07:
> Adding a 3d editor would be very useful if you have a properly designed 3d MAP. If you are in an indoor environment using a 3D edit would be easier, and will help avoiding collisions during the [...]

The image that i provide here was an idea to how to move the items in the flight plan. Maybe adding arrows to contol de X,Y,Z axis of the elements
.

omid-esrafilian, 2024-03-07:
Indeed, it would be useful to be able to edit the mission in the 3D view or in 3D in general. I had the same thoughts, as it would be more convenient. 
This could also be considered as an additional feature in the future.
Thanks for the suggestion.

closed 2026-10-04 by github-actions[bot]