mavlink/qgroundcontrol #11204: Legacy config still in QGC v4.3.0 causing problems

SeptentrioGNSS, 2024-03-06:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Under application settings in QGC there is a toggle for auto config of devices like RTK GPS. This means when connecting a RTCM capable receiver straight to QGC this should house a config that a Septentrio mosiac Go will be configured to become a base station. This is all fine if a new QGC stable release will also include an updated GPSDriver as this is currently using an older version of this repository. 

## Current Behavior
When connecting a RTK GPS receiver the ground control station software has toggles to auto configure these GNSS receivers to use the RTCM date provided by them. In the current and latest stable version of QGC, when connecting a Septentrio mosaic Go Heading module over USB to the software a legacy config is uploaded **continuously**. When for example you just want to test the configuration of the base station capable module before flight the parameters that you configure are sent on startup of the Autopilot HW but are immediately overwritten by the continuous config from a legacy parameter called upon by old GPSDrivers

## Steps to Reproduce:
1. Connect Septentrio mosaic Go heading module over USB to the latest stable version of QGC
2. have the autoconfigure RTK GPS toggle on (in application settings)
3. Power Autopilot (e.g. Holybro Pixhawk 6c) over USB, mosaic module is powered over USB, both are connected to computer running QGC
4. Double check the GNSS receiver configuration in RxControl or via WebUI for the Septentrio receiver and see the config defined by the macro GPS_CONFIG_RTCM in the GPS driver

## System Information
When posting bug reports, include the following information
- Operating System: Windows10
- QGC Version: v4.3.0
- QGC build: stable
- Flight Controller: Pixhawk 6c
- Autopilot (with version): PX4 v1.14.0

## Detailed Description
Long ago, this macro was called upon in the GPSDriver repository that is utilized in both PX4 as well as in the QGC build. This has been removed but the current stable version of the GPSDriver is still using an older version where the macro is called. The call has been cleaned up in the main but the macro is still in place everywhere. Having this removed will better allign with current development for the base configuration for Septentrio receivers

## Log Files and Screenshots
- GPSDriver: @ f48cc01 (version used in the build for the 4.3.0 stable release of QGC)
- Commit related to the macro: GPSDrivers [624a9b6](https://github.com/PX4/PX4-GPSDrivers/commit/624a9) 
- macro in question: 
[image]

SeptentrioGNSS, 2024-03-06:
More clarification can be provided if needed.

FYI @flyingthingsintothings

DonLakeFlyer, 2024-03-06:
@bkueng Any ideas here?

SeptentrioGNSS, 2024-03-07:
re-evaluated for further steps to take place.