mavlink/qgroundcontrol #11205: Custom Joystick actions on Android using JoystickMavCommands.json

ivanzivotic, 2024-03-06:
## Expected Behavior
For joystick control, there is an option for defining custom Mavlink messages that can later be assigned for the joystick buttons, enabling the user to send custom messages to the vehicle on button click. As I saw in the code that these message definitions are read from the file named JoystickMavCommands.json (line 120 of current master branch in Joystick.cc file), I was able to define custom messages on Windows and Ubuntu OS by putting the file to the working directory from which QGC is being started. However, this is not possible to do on Android as the process of starting an application (QGC in this case) is not done that way. Therefore, I am not sure where should the JSON file be located and if the QGC application even has the permissions for reading it.

## Current Behavior
Currently, there is no way to add this file on Android.

## System Information
When posting bug reports, include the following information
- Operating System: Android 14 (Samsung OneUI 6.0)
- QGC Version: 4.3.0
- QGC build: stable
- Autopilot (with version): Ardupilot Copter 4.3.5

Maybe the fix for this issue would be to set a static path from which the application will try to read the file and instruct users to always use that exact path for configuring these custom messages.

DonLakeFlyer, 2024-03-06:
This file should be coming from somewhere in the regular Documents/QGroundControl locations not the app root.

DonLakeFlyer, 2024-03-06:
Also double-check where CustomActions.json is loaded from.

DonLakeFlyer, 2024-03-06:
The "Enable Custom Actions" settings seems to have been lost in the conversion to new UI style. Need to add it back. There is a CustomActions directory where these should be loaded from.

DonLakeFlyer, 2025-11-25:
> The settings is missing and needs to be added back before this will work

No, that is wrong. The setting to select joystick actions is under Fly View Settings, which is strange but it it what it is for now. 

The problem is that on Android 10+ the file location being used isn't available when you don't have an SD card is not visible through My Files. So for 5.0 (for now) the solution is to always use an SD Card on your android device (General Settings). Which then gives you access to the MavlinkAction dir.