mavlink/qgroundcontrol #11206: SEGMENTATION FAULT ERROR ON QGC STARTUP

nvnanil, 2024-03-06:
I am using Ubuntu 22.04 for running QGC (Ver 4.3). After connecting it to my VOXL 2 drone network and launching QGC it crashes after a few seconds of startup with the following error:
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/nvnanil/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/nvnanil/Documents/QGroundControl/Logs"
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
Ignoring ssl certificates due to OpenSSL version mismatch
Map Cache in: "/home/nvnanil/.cache/QGCMapCache300" / "qgcMapCache.db"
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
Download QGC stable version failed "Error during download. Error: 3"
Adding target QHostAddress("192.168.8.1") 14550
Ignoring ssl certificates due to OpenSSL version mismatch
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
Ignoring ssl certificates due to OpenSSL version mismatch
Ignoring ssl certificates due to OpenSSL version mismatch
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters cache match failed /home/nvnanil/.config/QGroundControl.org/ParamCache/1_1.v2
QCoreApplication::postEvent: Unexpected null receiver
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum

(QGroundControl:84330): GStreamer-CRITICAL **: 16:26:07.610: gst_caps_from_string: assertion 'string' failed

(QGroundControl:84330): GStreamer-CRITICAL **: 16:26:07.610: gst_pad_template_new: assertion 'caps != NULL' failed

(QGroundControl:84330): GStreamer-CRITICAL **: 16:26:07.610: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(QGroundControl:84330): GStreamer-CRITICAL **: 16:26:07.610: gst_element_class_add_pad_template: assertion 'GST_IS_PAD_TEMPLATE (templ)' failed

(QGroundControl:84330): GStreamer-Video-CRITICAL **: 16:26:07.610: gst_video_decoder_init: assertion 'pad_template != NULL' failed

(QGroundControl:84330): GStreamer-WARNING **: 16:26:07.610: Element vaapidecode0 has an ALWAYS template src, but no pad of the same name

(QGroundControl:84330): GStreamer-CRITICAL **: 16:26:07.610: gst_pad_link_full: assertion 'GST_IS_PAD (sinkpad)' failed
Segmentation fault (core dumped)
```
I have downloaded the QGroundControl.AppImage file to my Downloads folder and am running it with the terminal command ``` ```./QGroundControl.App Image``` The version works perfectly with jMAVSim.

Any thoughts or solutions would be helpful.

DonLakeFlyer, 2024-03-07:
Can you get a stack trace from the crash?

nvnanil, 2024-03-07:
execve("./QGroundControl_4_3.AppImage", ["./QGroundControl_4_3.AppImage"], 0x7fffe5326db0 /* 62 vars */) = 0
brk(NULL)                               = 0x1c83000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffee40cbd70) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f1a744ac000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/glibc-hwcaps/x86-64-v3/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/glibc-hwcaps/x86-64-v3", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/glibc-hwcaps/x86-64-v2/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/glibc-hwcaps/x86-64-v2", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/tls/haswell/x86_64/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/tls/haswell/x86_64", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/tls/haswell/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/tls/haswell", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/tls/x86_64/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/tls/x86_64", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/tls/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/tls", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/haswell/x86_64/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/haswell/x86_64", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/haswell/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/haswell", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/x86_64/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/x86_64", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_sensors/lib", {st_mode=S_IFDIR|0775, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/glibc-hwcaps/x86-64-v3/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/glibc-hwcaps/x86-64-v3", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/glibc-hwcaps/x86-64-v2/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/glibc-hwcaps/x86-64-v2", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/tls/haswell/x86_64/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/tls/haswell/x86_64", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/tls/haswell/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/tls/haswell", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/tls/x86_64/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/tls/x86_64", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/tls/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/tls", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/haswell/x86_64/libdl.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/nvnanil/ariac_ws/install/ariac_plugins/lib/haswell/x86_64", 0x7ffee40caf90, 0) = -1 ENOENT (No such file or directory)
[... cut ...]
wait4(-1, [{WIFSIGNALED(s) && WTERMSIG(s) == SIGSEGV && WCOREDUMP(s)}], 0, NULL) = 8514
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_KILLED, si_pid=8514, si_uid=1000, si_status=SIGSEGV, si_utime=121, si_stime=94} ---
rt_sigreturn({mask=[]})                 = 8514
write(2, "Segmentation fault (core dumped)"..., 33) = 33
wait4(-1, 0x7fffbcabac8c, WNOHANG, NULL) = -1 ECHILD (No child processes)
read(10, "", 8192)                      = 0
exit_group(139)                         = ?
+++ exited with 139 +++

DonLakeFlyer, 2024-03-09:
I'm not sure what that is but it isn't a stack trace. The gdb command for stack trace is 'bt'

nvnanil, 2024-03-09:
I tried running the file with ```gdb run``` but it showed the message 
"No stack trace"

nvnanil, 2024-03-18:
@DonLakeFlyer Any help in solving this issue?

DonLakeFlyer, 2024-04-05:
You need to:
* `gdb QGroundControl.AppImage`
* At the gdb prompt type `run`
* When it crashes and the gdb prompt type `bt 100`

ascharlton, 2025-02-01:
I believe Im having the same issue and going in circles. Im running physical machine Ubuntu 22.04 and QGC 4.4.3 (2 Dec 2024 release). Im trying to get rtsp video feeds working. 

Im providing a stable rtsp stream from a raspberry pi and Siyi A8 camera and re-streaming via mediamtx (Ive tried a few other rtsp servers also etc ) over wifi. I can load QGC with video streaming disabled or enabled as long as there is no video feed running from the source. Soon as I start a video feed it QGC crashes (core dump).

If I remove gstreamer1.0-vaapi I can get the video feed to hold and is stable - until I connect a drone (in my case HolybroV6X) then it crashes the video ouput. If I put vaapi back then I get QGC crashing again.

Happy to provide crash dumps --logging:full etc... but because of the different combinations I dont know where to start.

Can someone help me on this please??

ascharlton, 2025-02-05:
It seems I may be struggling with an inferior nvidia card I suspect.  Im getting lots of these errors in DMESG when the video feed is detected 👍 
[31780.254192] nouveau 0000:01:00.0: Direct firmware load for nouveau/nvc1_fuc084d failed with error -2

I have a Nvidia GF 108  which doesnt have good drivers for recent kernels so I will see if a kernel downgrade will help.

ascharlton, 2025-02-06:
I think this was due my original Ubuntu 22.04 install or more specifically the video card drivers on my machine, QGC 4.4.3 daily works fine on a live version of Ubuntu 24 for example. Back to the drawing board, at least I have a place to start.

closed 2025-05-03 by HTRamsey