mavlink/qgroundcontrol #11236: Windows Flags Latest CI Build as a Virus

AlexKlimaj, 2024-03-11:
Looks like Windows daily downloaded from the Docs doesn't have video streaming or the mavlink console working.
But downloading the daily from CI gets flagged as a virus.

[image]

zdanek, 2024-03-11:
interesting, can you check what causes this to be marked as an virus?
IMO this is to be checked, but I see no info about your antivir software. Can you help?
Maybe some particular library, that's a part of an install file, is considered as "harmful"? I'd bet it's based on heuristics and it's a false positive, but let's know what are the reasons.

AlexKlimaj, 2024-03-11:
Trojan:Script/Wacatac.B!ml

[image]

HTRamsey, 2024-03-11:
it's because it's a .exe.zip for some reason

HTRamsey, 2024-03-11:
Yeah I just doubled checked and the exact same file downloaded as just a .exe from [here](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/releases/daily_builds.html) is fine. so it is the extension

AlexKlimaj, 2024-03-11:
Are you sure that's the same file? Installed from that file and it shows a build date of Feb 8 2024. Commit 01ccf028

[image]

HTRamsey, 2024-03-11:
Oh I guess the dailys aren't being updated there or something. I have no idea how that is set up. But you can go back several months and download old CI releases and they all throw a virus warning, most likely because of the extension.

DonLakeFlyer, 2024-04-05:
So this is the zips from CI artifacts right? Not much I can do about that. Does regular doc daily build download link work? That's a raw .EXE

AlexKlimaj, 2024-04-07:
Yeah. However it looks like the latest builds from master are working again. Not sure why windows thought they were viruses for a while.

closed 2024-04-07 by AlexKlimaj