mavlink/qgroundcontrol #11252: The latest QGroundControl and the latest PX4 firmware parameters do not match.

dagouxiong233, 2024-03-14:
## Current Behavior
1:FW_ARSP_MODE, 1:CBRK_AIRSPD_CHK
[image]

## System Information
- Operating System: ubuntu and windows
- QGC Version: v4.3.0
- Flight Controller:V2.4.8
- Autopilot (with version): PX4 [Stable Release v1.14.0]

dagouxiong233, 2024-03-14:
In the latest version of the PX4 firmware, there have been changes to some parameters. The parameter FW_ARSP_MODE has been renamed to FW_USE_AIRSPD, and the CBRK_AIRSPD_CHK parameter has been replaced with SYS_HAS_NUM_ASPD. For operations without an airspeed sensor, you would set SYS_HAS_NUM_ASPD to 0. Similarly, if you do not wish to use the airspeed sensor data in the controller, you would set FW_USE_AIRSPD to 0​

dagouxiong233, 2024-03-14:
https://docs.px4.io/main/en/releases/main.html

DonLakeFlyer, 2024-04-05:
This is going to require version check on parameter access for forward/back compatibility

sfuhrer, 2024-04-08:
QGC shouldn't have to worry about `FW_USE_AIRSPD`. What is the info you need to have in QGC, is it whether there is an airspeed sensor present on the vehicle that needs calibration? How do we do it for the other senors (accel, gyro, mag), do we just assume at least one of each is always present? `SYS_HAS_NUM_ASPD` could be used for that, but can we also consider relying on mavlink instead? Ideally there would be a message from the autopilot about which sensors are present (or, even better, which ones require calibration). For the meantime, there are already existing mavlink messages that can be checked to see if there is any airspeed sensor present at all, e.g. [SCALED_PRESSURE](https://mavlink.io/en/messages/common.html#SCALED_PRESSURE2) has a differential pressure field, or the airspeed field of [VFR_HUD](https://mavlink.io/en/messages/common.html#VFR_HUD).

DonLakeFlyer, 2024-04-08:
> What is the info you need to have in QGC, is it whether there is an airspeed sensor present on the vehicle that needs calibration?

Correct.

> For the meantime, there are already existing mavlink messages that can be checked to see if there is any airspeed sensor present at all, e.g. [...]

But what happens if the airspeed sensor hasn't been calibrated yet. Will that still work?

sfuhrer, 2024-04-11:
>But what happens if the airspeed sensor hasn't been calibrated yet. Will that still work?

Currently yes, though we're working on enabling multi airspeed sensors which would then require an adapted workflow in QGC again. 
@dagar have you already given this some thoughts?

DenysZaytsev, 2024-04-30:
Hi guys, 
I'm trying to build a bootloader and a firmware for Matek H743-WING v3 (using make matek_h743_bootloader and make matek_h743_default from the latest release 1.14). After I upload it to the FC, I see those messages in QGC 
[image]
I tried to disable AirSpeed sensor by disabling parameters
[image]
and
[image]
I tried with VTOL and standard plane with no luck. 

Due to this issue I can't calibrate AirFlow sensor - it just fails and I can't also disable it completely.
Also I can't Arm the vehicle, because preflight checks are not passed:
[image]
[image]
[image]
[16:49:07.451] Critical: Arming denied: Resolve system health failures first

Could you pls let me know how to fix the issue either without AirFlow sensor (I was up to use Matek ASPD4525) or to make that one work?
Thanks in advance.

DonLakeFlyer, 2024-04-30:
> what do you do for the Mag calibration in QGC to figure out when to display it as "mag needs calibration"?

@sfuhrer 
```
bool SensorsComponent::setupComplete(void) const
{
    foreach (const QString &triggerParam, _deviceIds) {
        if (_vehicle->parameterManager()->getParameter(FactSystem::defaultComponentId, triggerParam)->rawValue().toFloat() == 0.0f) {
            return false;
        }
    }
    bool magEnabled = true;
    if (_vehicle->parameterManager()->parameterExists(FactSystem::defaultComponentId, _magEnabledParam)) {
        magEnabled = _vehicle->parameterManager()->getParameter(FactSystem::defaultComponentId, _magEnabledParam)->rawValue().toBool();
    }

    if (magEnabled && _vehicle->parameterManager()->getParameter(FactSystem::defaultComponentId, _magCalParam)->rawValue().toFloat() == 0.0f) {
        return false;
    }

    if (_vehicle->fixedWing() || _vehicle->vtol() || _vehicle->airship()) {
        if (!_vehicle->parameterManager()->getParameter(FactSystem::defaultComponentId, _airspeedDisabledParam)->rawValue().toBool() &&
                _vehicle->parameterManager()->getParameter(FactSystem::defaultComponentId, _airspeedBreakerParam)->rawValue().toInt() != 162128 &&
                _vehicle->parameterManager()->getParameter(FactSystem::defaultComponentId, _airspeedCalParam)->rawValue().toFloat() == 0.0f) {
            return false;
        }
    }

    return true;
}
```

DonLakeFlyer, 2024-04-30:
@dagouxiong233 @sfuhrer Can you both review the changes in QGC to makes sure I have this version specific handling correct? Specifically these two methods:
* `bool SensorsComponent::_airspeedCalSupported  (void) const`
* `bool SensorsComponent::_airspeedCalRequired   (void) const`

DonLakeFlyer, 2024-04-30:
@DenysZaytsev Once CI finishes can you test this? You can pull the install artifacts from the GitHub Actions.

DenysZaytsev, 2024-04-30:
> @DenysZaytsev Once CI finishes can you test this? You can pull the install artifacts from the GitHub Actions.

Yes, I can test with Matek ASPD4525
So far I just compiled from release branch 1.14 (previously I was compiling from the main branch) and there are no errors in QGC (cause no changes yet been made in that release), but AirSpeed calibration still doesn't work, however at least I disabled it completely and now can arm.

DonLakeFlyer, 2024-05-01:
@DenysZaytsev Builds are ready. What OS do you use? I can point you to the install.

PR #11425 (2024-06-11): Rework PX4 airspeed cal due to parameter name/functionality changes
Fix #11252
closed 2024-06-11 by DonLakeFlyer