mavlink/qgroundcontrol #11270: qmlglsink is not correctly built with cmake in Android case.

hitagisol, 2024-03-26:
I'm trying to build the Android version in the current master([c09fd99](https://github.com/mavlink/qgroundcontrol/commit/c09fd991ef64ab3c6660b041f8fc5f207066f282)) with cmake. The qmlglsink lib is not built in this case. 
By looking at the qmlglsink's [CMakeLists.txt](https://github.com/mavlink/qgroundcontrol/blob/c09fd991ef64ab3c6660b041f8fc5f207066f282/libs/qmlglsink/CMakeLists.txt), I notice that the following module check is bypassed in Android case: `pkg_check_modules(GST
            ${GST_DEPENDENCIES}
        )` but then the Android relates definitions `target_compile_definitions(qmlglsink PUBLIC HAVE_QT_ANDROID)` is applied only if GST_FOUND is true, which will be always false for Android.

HTRamsey, 2024-03-26:
#11258

hitagisol, 2024-03-26:
> #11258

Cool! Sorry I didn't notice that commit. Thanks!

closed 2024-03-26 by hitagisol