mavlink/qgroundcontrol #11272: QGC AppImage does not include dynamic library for QTQuickControls

TOTON95, 2024-03-26:
## Expected Behavior
The program should launch.

## Current Behavior
Launch is stuck at the following message:

```bash
QQmlApplicationEngine failed to load component
qrc:/qml/MainRootWindow.qml:11:1: module "QtQuick.Controls" version 6.6 cannot be imported because:
Cannot load library /tmp/.mount_QGroungsoA6b/Qt/qml/QtQuick/Controls/Basic/libqtquickcontrols2basicstyleplugin.so: (libQt6QuickControls2Basic.so.6: cannot open shared object file: No such file or directory)
```

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Launch the program as usual through a terminal
2. Observe output

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04 LTS
- QGC Version: Master
- QGC build: Daily
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description

This is the AppImage that was subject to test from commit [a95dfd1](https://github.com/mavlink/qgroundcontrol/commit/a95dfd1160b4cecc4db479b9df5d49f66fd452ec): 
https://github.com/mavlink/qgroundcontrol/actions/runs/8439025593/artifacts/1360125262

These are the libraries that can be found in the AppImage running 

`./QGroundControl.AppImage --appimage-mount` 

and in another terminal

`ldd QGroundControl`

at the `tmp` directory shown by the first command.

Output:

```bash
linux-vdso.so.1 (0x00007fffca6d0000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fdb88f8b000)
	libSDL2-2.0.so.0 => /lib/x86_64-linux-gnu/libSDL2-2.0.so.0 (0x00007fdb88e36000)
	libgstgl-1.0.so.0 => /lib/x86_64-linux-gnu/libgstgl-1.0.so.0 (0x00007fdb88dbe000)
	libgstvideo-1.0.so.0 => /lib/x86_64-linux-gnu/libgstvideo-1.0.so.0 (0x00007fdb88d0d000)
	libgstbase-1.0.so.0 => /lib/x86_64-linux-gnu/libgstbase-1.0.so.0 (0x00007fdb88c8f000)
	libgstreamer-1.0.so.0 => /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0 (0x00007fdb88b46000)
	libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fdb88ae6000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fdb889bc000)
	libQt6Location.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Location.so.6 (0x00007fdb887cf000)
	libQt6QuickControls2.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6QuickControls2.so.6 (0x00007fdb887b8000)
	libQt6Quick3D.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Quick3D.so.6 (0x00007fdb8864b000)
	libQt6Quick3DRuntimeRender.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Quick3DRuntimeRender.so.6 (0x00007fdb882d1000)
	libQt6Quick.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Quick.so.6 (0x00007fdb87c34000)
	libQt6Charts.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Charts.so.6 (0x00007fdb87a19000)
	libQt6Widgets.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Widgets.so.6 (0x00007fdb872e0000)
	libQt6TextToSpeech.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6TextToSpeech.so.6 (0x00007fdb872c0000)
	libQt6Multimedia.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Multimedia.so.6 (0x00007fdb871cd000)
	libQt6Gui.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Gui.so.6 (0x00007fdb867f3000)
	libQt6Positioning.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Positioning.so.6 (0x00007fdb86757000)
	libQt6Qml.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Qml.so.6 (0x00007fdb861a3000)
	libQt6Network.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Network.so.6 (0x00007fdb85ffa000)
	libQt6Sql.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Sql.so.6 (0x00007fdb85fa9000)
	libQt6Xml.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Xml.so.6 (0x00007fdb85f7e000)
	libQt6Core5Compat.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Core5Compat.so.6 (0x00007fdb85f1f000)
	libQt6SerialPort.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6SerialPort.so.6 (0x00007fdb85efc000)
	libQt6Test.so.6 => /tmp/.mount_QGroungsoA6b/./Qt/lib/libQt6Test.so.6 (0x00007fdb85e99000)
[... 66 lines cut ...]
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fdb812bb000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007fdb811dc000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007fdb811ab000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007fdb811a4000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007fdb81195000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007fdb810e6000)
	libwrap.so.0 => /lib/x86_64-linux-gnu/libwrap.so.0 (0x00007fdb810d8000)
	libsndfile.so.1 => /lib/x86_64-linux-gnu/libsndfile.so.1 (0x00007fdb81059000)
	libasyncns.so.0 => /lib/x86_64-linux-gnu/libasyncns.so.0 (0x00007fdb80e53000)
	libapparmor.so.1 => /lib/x86_64-linux-gnu/libapparmor.so.1 (0x00007fdb80e3e000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fdb80e24000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007fdb80e1b000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fdb80df2000)
	liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007fdb80dd1000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007fdb80cb3000)
	libnsl.so.1 => /lib/x86_64-linux-gnu/libnsl.so.1 (0x00007fdb80c96000)
	libFLAC.so.8 => /lib/x86_64-linux-gnu/libFLAC.so.8 (0x00007fdb80c56000)
	libogg.so.0 => /lib/x86_64-linux-gnu/libogg.so.0 (0x00007fdb80c49000)
	libvorbis.so.0 => /lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007fdb80c1b000)
	libvorbisenc.so.2 => /lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007fdb80b70000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fdb80b4d000)
```

## Log Files and Screenshots
N/A

HTRamsey, 2024-04-05:
@TOTON95 Are we good here now?

TOTON95, 2024-04-05:
@HTRamsey Indeed this should be fixed by #11273. Hopefully, the cmake implementation will solve these problems, until then I can still check once in a while for builds made from qmake. Closed, thank you!

closed 2024-04-05 by TOTON95

drewskoots, 2024-06-06:
It seems this libraries weren't added to the build container Dockerfile, when I try to compile there it throws the error `cp: cannot stat '/opt/Qt/6.6.1/gcc_64/lib/libQt6QuickControls2Basic.so.6': No such file or directory`

HTRamsey, 2024-06-06:
@drewskoots that's due to the Qt version. You need 6.6.3 or else that library is missing. The current Dockerfile should install 6.6.3.

drewskoots, 2024-06-07:
Ah yes thanks! I was trying to build the AirspeedCal branch which still had version 6.6.1 in the dockerfile.

Thanks!

Andrew Madden