mavlink/qgroundcontrol #11284: Video lag

darshan-crypto, 2024-04-05:
## Expected Behavior
I expected the video feed to stream smoothly and without any significant delay.

## Current Behavior
The video feed is lagging consistently, with a delay of several seconds between real-world events and their appearance in the video stream.

## System Information
When posting bug reports, include the following information
- Operating System: Android
- QGC Version: v4.3.0
- QGC build: stable

## Video

https://github.com/mavlink/qgroundcontrol/assets/76841538/c2de50ce-05dc-4b3a-88a1-f7690b4e4f4a

HTRamsey, 2024-05-01:
This will likely be better in v5.0

ericjohnson97, 2024-07-14:
As another data point. I also have noticed my streams become more laggy and choppy recently recently. 

Here is a video from version v4.4.0 

https://github.com/user-attachments/assets/2902efba-1b28-4766-81a4-dfda582cf955

Here is a clip after I downgraded to v4.2.7

https://github.com/user-attachments/assets/daa9f13b-1af4-434d-ade4-83c3fb130815

Adding this as a data point. Not sure what has changed in the last year, but I hope it gets better in v5.0! When is v5.0 supposed to be released? Also, If I build the project from source, would I be able to get the v5 improvements now?

HTRamsey, 2024-07-14:
Yeah the daily is running gstreamer 1.22.12 vs 1.18.5 in the older versions. Also, I have been working on adding an alternative QtMultimedia based video receiver that uses ffmpeg instead

ericjohnson97, 2024-07-15:
I got curious and downloaded the apk from here
https://github.com/mavlink/qgroundcontrol/actions/runs/9929355840

I can confirm the video is much better. Also it looks like there are big changes coming to QGC in the next release. whoa!

Since there seems to be a lot of changes in this build from the last release. I'll probably stick with running 4.2.7 for the time being just to be safe. I would be interested in helping test the next version when the time comes.  Where is the best place to stay in the loop on development?

closed 2024-10-16 by HTRamsey