mavlink/qgroundcontrol #11293: QGC Can not detect VTOL frame for Plan VTOL Mission with Ardupilot code

moreba1, 2024-04-05:
Hi

I use fixed wing VTOL code of Ardupilot , but when drone is connected to software , QGC in Plan view can not detect that it is a VTOL drone Not Fixedwing Drone . So QGC Detect a Fixed wing and VTOL Take off and Landing Can not be choice and used.

that2, 2024-04-20:
Same problem over here. Using Arduplane 4.4.4 With Q_Enable for VTOL capabilities but when connected to QGC it shows vehicle type as Fixed wing so no options for VTOL Tkoff and landing.

that2, 2024-04-20:
@moreba1 try changing the Q_MAV_TYPE to 20, default is auto. That worked for me.

moreba1, 2024-04-25:
> @moreba1 try changing the Q_MAV_TYPE to 20, default is auto. That worked for me.

Thanks , yes IT works. I had set Q_MAV_TYPE  to auto.

https://github.com/ArduPilot/ardupilot/issues/7137

DonLakeFlyer, 2024-04-25:
Closing since this is resolved.

closed 2024-04-25 by DonLakeFlyer