mavlink/qgroundcontrol #11295: Restore iOS Builds

HTRamsey, 2024-04-05:
Doesn't seem like too big of an issue to get these back up, but I don't personally have a way to test it.

HTRamsey, 2024-04-30:
@DonLakeFlyer What are your thoughts on this? Is this something anybody has been asking for?

DonLakeFlyer, 2024-05-01:
The problem is you can build something. But there is no realistic way to get it onto an iPhone the way the phone is locked down. Someone would need to setup up their own beta dev program in apple play. And then I think you can get an app onto a phone which didn't come through the store. But you need to put the phone on a dev mode as well if I remember. All of which is nothing a normal user will ever do. You can't just stuff things onto an iPhone like you can with Android.

The only useful case would be if an iOs build went directly from CI to Apple Play. There used to be a guy who would hand carry stable releases to the store and release them. He's gone now and I never bother with trying to do it manually. There hasn't been an iOs store update of QGC in 5-6 years.

DonLakeFlyer, 2024-05-01:
As far as commercial folks. I only know of Sentera that had an iOS custom build they released. But they did that ll on their own.

HTRamsey, 2024-05-01:
Isn't that changing as of ios17? I think the EU is pushing them to allow sideloading apps.

HTRamsey, 2024-05-01:
Taken from Reddit:

"The point is that installing 3rd party apps, so called 'sideloading', currently has a few restrictions, such as requiring a developer account (free or paid) and the installed apps being effectively linked to your account.

Additionally you have to reinstall once a week (free) or year (paid), the apps are not permanently installed.

With the new EU regulations, all those downsides are supposed to disappear and 'sharing apps' should theoretically be as easy as simply sending someone an IPA-file, which the other guy then can freely install on-device.

IOS 17.0.x currently does not contain any changes in regards how 3rd party app installation works, but since those new regulations become mandatory as of March 2024, it is expected to be included in a future update (likely 17.2 or 17.3)."

So maybe it's not important unless people ask for it, but if we can at least provide a working CI workflow that builds it, it might be worth mentioning in the docs.

DonLakeFlyer, 2024-05-01:
Ah, interesting. Certainly can't hurt to have CI iOS then.

HTRamsey, 2024-05-31:
Closing since we have a usable CI workflow now.

closed 2024-05-31 by HTRamsey