mavlink/qgroundcontrol #11298: App Settings icons not colored correctly

DonLakeFlyer, 2024-04-05:
[image]

I'm guessing caused by recent changes to Qt5 Compat qml stuff and QGCColoredImage

DonLakeFlyer, 2024-04-05:
@HTRamsey Could you look at this? Let me know if you need me to poke at it as well.

DonLakeFlyer, 2024-04-05:
More examples:
[image]

The gear icon should be white.

DonLakeFlyer, 2024-04-05:
This is on a Mac

HTRamsey, 2024-04-06:
So this one is kinda stupid and I'm not sure why it has to be different than the old ColorOverlay: https://forum.qt.io/topic/144070/qt6-color-svg-using-multieffect/2

s-lisovenko, 2024-04-07:
Unfortunately there is no simple way to replace ColorOverlay in Qt6. I think we should continue use Qt5Compat for this. 
https://github.com/mavlink/qgroundcontrol/pull/11310

HTRamsey, 2024-04-08:
Yeah looks like you'd have to change the fill color of every SVG to white then color it in QML

closed 2024-06-02 by HTRamsey