mavlink/qgroundcontrol #11313: how to view a single qml view?

shenshiqiSSQ, 2024-04-08:
Hello, may I ask how to view a single qml view? When I open the qml view through qtcreator, various errors will be reported, resulting in a failure to open QML

[image]
[image]
[image]

HTRamsey, 2024-04-29:
This is a question for Qt

closed 2024-04-29 by HTRamsey

DonLakeFlyer, 2024-04-29:
You can't edit QGC Qml in the Qt Designer. You have to just use a text editor.