mavlink/qgroundcontrol #11333: No elevation data for territories surrounding Tbilisi Georgia (the country)

someone-ua, 2024-04-12:
----

## Expected Behavior
Elevation data is available for territories surrounding Tbilisi city as for other territories. Requests to terrain-ce.suite.auterion.com returns valid elevation data.

## Current Behavior
Elevation data is not available for territories surrounding Tbilisi city. Requests to terrain-ce.suite.auterion.com returns HTTP error 500 with the following  JSON data
{"error":"tile not found"}

## Steps to Reproduce:
1. Create new mission.
2. Set altitude mode to Calc Above Terrain
3. Place Takeoff point and a Waypoint near Tbilisi city
4. Open the following link in browser https://terrain-ce.suite.auterion.com/api/v1/carpet?points=41.68,43.99,41.69,44

## System Information
When posting bug reports, include the following information
- Operating System: MacOs 14.2.1 (23C71)
- QGC Version: 4.3.0
- QGC build: stable
- Flight Controller: not used
- Autopilot (with version): not used

## Detailed Description
I've found that currently elevation data is not available for territories surrounding Tbilisi city in Georgia (the country), however it was available before and is still available for other territories in Georgia.
According to my investigation, terrain-ce.suite.auterion.com currently returns "tile not found" for that territories. For example https://terrain-ce.suite.auterion.com/api/v1/carpet?points=41.68,43.99,41.69,44
It seems there is no possibility to use alternative sources of elevation data.

## Log Files and Screenshots
[image]
[image]
[image]
[qgroundcontrollog.txt](https://github.com/mavlink/qgroundcontrol/files/14960047/qgroundcontrollog.txt)

mj-saunders, 2025-10-24:
I'm having the same problem with tiles from a different part of the world. Currently I'm looking into how much work might be involved in adding another elevation data provider.
Here's an example of a working tile response from auterion:  https://terrain-ce.suite.auterion.com/api/v1/carpet?points=51.14,-2.26,51.15,-2.25
I'll have a look to see if the source or docs clarify, but do we at least know what data format this is, and if it's something standard we might expect frm other services?

mj-saunders, 2025-10-24:
Looking at this PR and discussion (https://github.com/mavlink/qgroundcontrol/pull/10740), I'm not certain but maybe the auterion dataset is coming originally via opentopography.

I've checked and their API has data for e.g. the Tblisi region in question:
```sh
$ curl -X 'GET'   'https://portal.opentopography.org/API/globaldem?demtype=COP30&south=41.68&north=41.69&west=43.99&east=44&outputFormat=AAIGrid&API_Key=demoapikeyot2022
```

The format chosen here closely resembles that from the auterion service, so I'm hoping it should involve minimal manipulation to be able to use much of the existing code in `qgroundcontrol`.

It's not a fast service, and requires an API key. But there's precedent for the latter in existing `QGC` code, so I remain hopeful.
At the very least we could perhaps integrate it as an offline maps service, purely for elevation data.

closed 2025-10-24 by DonLakeFlyer