mavlink/qgroundcontrol #11347: Warn user controller battery low!

woutertollenaar, 2024-04-17:
**Read the battery status of the Android controller and alert the user if the battery is running low.
This prevents an uncontrollable drone due to an empty controller.**

**Tell us a bit about the feature:**
- What problem does it solve?
It ensures that the user does not start piloting his drone when his controller is running low.
- What does it do?
it reads the android battery status API and alerts the user when the controller is low. 
- What flight stacks must it work with? (All, PX4, ArduPilot)
All
- Are there other systems that have this feature?
the Qgroundcontroll app has a feature to alert the user when the drone battery is low
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
read API

HTRamsey, 2024-04-17:
Good references for this:
[Java](https://github.com/FalsinSoft/QtAndroidTools/blob/master/QtAndroidTools/src/com/falsinsoft/qtandroidtools/AndroidBatteryState.java)
[Cpp](https://github.com/FalsinSoft/QtAndroidTools/blob/master/QtAndroidTools/QAndroidBatteryState.cpp)

Publicano, 2024-06-21:
Now try implementing one of the following solutions. 
https://falsinsoft.blogspot.com/2017/06/qml-get-android-battery-level-and-status.html
https://github.com/FalsinSoft/QtExamples/tree/master/QML/AndroidBatteryLevel

Does anyone already have it working?