mavlink/qgroundcontrol #11352: An error occurs when compiling QGC for Android system

BreederBai, 2024-04-18:
Compilation environment: windows11
Qt version: 6.6.1
QGC version: Today’s master
Qt configuration is as shown below：
[image]
The error is reported as shown below：
[image]

BreederBai, 2024-04-18:
When I try to use JDK11, Qt will report the following error：
[image]

HTRamsey, 2024-04-18:
For for first one, look at #11169. I don't know why some people have this problem. For second, just get a newer jdk version. It is required to support newer android versions.

BreederBai, 2024-04-24:
@HTRamsey I seem to have found the reason why I need to manually create the ANDROID-PACKAGE-SOURCE-DIR folder. It seems that there is an issue with the following few sentences of the script
[image]

HTRamsey, 2024-04-24:
That might be what #9932 is meant to fix, not sure. @DonLakeFlyer did you want to still do something about this for v4.4?

DonLakeFlyer, 2024-04-24:
This is with android qmake build right? Does android cmake build work correctly with respect to the manifest gymnastics for a regular non-custom build ?

BreederBai, 2024-04-24:
> 这是用 android qmake 构建的吧？ android cmake 构建对于常规非自定义构建的清单体操是否正常工作？

Hello, how do I build using CMake? Does the current code support it?

DonLakeFlyer, 2024-04-24:
Yes, it does. If you look at the CI workflow: https://github.com/mavlink/qgroundcontrol/blob/master/.github/workflows/android.yml you'll see how it's done in CI. I don't think docs are yet up to date with the latest changes.

BreederBai, 2024-04-24:
> 是的，它确实。如果您查看 CI 工作流程：https://github.com/mavlink/qgroundcontrol/blob/master/.github/workflows/android.yml，您将看到它是如何在 CI 中完成的。我认为文档尚未更新最新的更改。

Sorry, I know nothing about CI. Do you have a tutorial on configuring cmake in Qt Creator

BreederBai, 2024-04-24:
> 您应该能够在我链接到的文件中看到 cmake 命令。它是如何工作的相当明显。应该有关于如何在 qt Creator 中使用 cmake 构建的 Qt 文档。我使用 Visual Code 并从那里进行操作，因此我不熟悉 Qt Creator 如何与 cmake 配合使用。

Can Visual Code still be used for construction? This is too cool! Do you have a tutorial on how to use Visual Code to build? I found out how to use Qt Creator on the Internet. I just need to select the CMakeLists.txt file when I open the project with Qt Creator. I am in the process of building, and everything seems normal at present.

BreederBai, 2024-04-24:
@DonLakeFlyer Thank you for your guidance. Although there were some warnings, the program has been successfully built and run on my hardware.Unfortunately, there is no RTSP option available for video retrieval in the app I built. What do I need to do to have RTSP?

`Success
Android package built successfully in 17.304 ms.
  -- It can now be run from the selected device/emulator.
  -- File: C:/Users/songs/source/repos/qgc/qgroundcontrol/build/Qt_6_6_3_Clang_arm64_v8a-Debug/android-build//build/outputs/apk/debug/android-build-debug.apk
Warning: QML import could not be resolved in any of the import paths: QtQuick.Controls.Windows
Warning: QML import could not be resolved in any of the import paths: QtQuick.Controls.macOS
Warning: QML import could not be resolved in any of the import paths: QtQuick.Controls.iOS
Warning: QML import could not be resolved in any of the import paths: QGroundControl
Warning: QML import could not be resolved in any of the import paths: QGroundControl.ScreenTools
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Controls
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Palette
Warning: QML import could not be resolved in any of the import paths: QGroundControl.FactSystem
Warning: QML import could not be resolved in any of the import paths: QGroundControl.FactControls
Warning: QML import could not be resolved in any of the import paths: QGroundControl.FlightDisplay
Warning: QML import could not be resolved in any of the import paths: QGroundControl.FlightMap
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Controllers
Warning: QML import could not be resolved in any of the import paths: Custom.Widgets
Warning: QML import could not be resolved in any of the import paths: QGroundControl.SettingsManager
Warning: QML import could not be resolved in any of the import paths: QGroundControl.ArduPilot
Warning: QML import could not be resolved in any of the import paths: QGroundControl.QGCPositionManager
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Vehicle
Warning: QML import could not be resolved in any of the import paths: QGroundControl.PX4
Warning: QML import could not be resolved in any of the import paths: QGroundControl.MultiVehicleManager
Warning: QML import could not be resolved in any of the import paths: MAVLink
Warning: QML import could not be resolved in any of the import paths: Viewer3D
Warning: QML import could not be resolved in any of the import paths: org.freedesktop.gstreamer.GLVideoItem
Warning: QML import could not be resolved in any of the import paths: QGroundControl.ShapeFileHelper
Warning: QML import could not be resolved in any of the import paths: QGroundControl.UTMSP
Warning: QML import could not be resolved in any of the import paths: QtQuick.Extras
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Templates
Warning: QML import could not be resolved in any of the import paths: QGroundControl.ScreenToolsController
Warning: QML import could not be resolved in any of the import paths: QGroundControl.QGCMapEngineManager
Warning: QML import could not be resolved in any of the import paths: QGroundControl.AutoPilotPlugin
Warning: QML import could not be resolved in any of the import paths: QGroundControl.Viewer3D
Warning: QML import could not be resolved in any of the import paths: Viewer3D.Models3D
Warning: QML import could not be resolved in any of the import paths: Viewer3D.Models3D.Drones
00:58:44: 进程“C:\Qt\6.6.3\mingw_64\bin\androiddeployqt.exe”正常退出。
00:58:44: 正在读取调试需要的文件。
00:58:44: 包部署：运行命令“C:\Users\songs\AppData\Local\Android\Sdk\platform-tools\adb.exe -s 3fe6e4b8 pull /system/bin/app_process64 C:\Users\songs\source\repos\qgc\qgroundcontrol\build\Qt_6_6_3_Clang_arm64_v8a-Debug\android-app-process\app_process”。
00:58:44: 包部署：运行命令“C:\Users\songs\AppData\Local\Android\Sdk\platform-tools\adb.exe -s 3fe6e4b8 pull /system/bin/linker64 C:\Users\songs\source\repos\qgc\qgroundcontrol\build\Qt_6_6_3_Clang_arm64_v8a-Debug\android-app-process\linker64”。
00:58:44: 包部署：运行命令“C:\Users\songs\AppData\Local\Android\Sdk\platform-tools\adb.exe -s 3fe6e4b8 pull /system/lib64/libc.so C:\Users\songs\source\repos\qgc\qgroundcontrol\build\Qt_6_6_3_Clang_arm64_v8a-Debug\android-app-process\libc.so”。
00:58:44: Elapsed time: 00:43.`

BreederBai, 2024-04-24:
> [这](https://github.com/mavlink/qgroundcontrol/blob/d7859938184df8867b658c12fc62a217ad468a62/.github/workflows/android.yml#L141)是 Android 中 gstreamer CI 的相关部分。我很快就会将该部分移至 cmake 中自动发生。

I have placed gstreamer in the root directory of qgroundcontrol and read cmakelists.txt. It seems that a Release version is required to enable gstreamer. I am not sure if my understanding is correct
[image]

HTRamsey, 2024-04-24:
No, it shouldn't need to be a Release build.

BreederBai, 2024-04-25:
> 不，它不需要是发布版本。

Hello, in CI, I only saw downloading and decompressing gstreamer, and did not see how it was added to the build. Is it because the program using CMake has not yet used gstreamer

HTRamsey, 2024-04-25:
That's all you have to do. CMake will find it if it is in the root source folder and has the correct name.

BreederBai, 2024-04-25:
> 这就是你所要做的。如果它位于根源文件夹中并且名称正确，CMake 将找到它。

Strangely, using cmake to build, without the gstreamer folder, it can still be successfully built, and it feels like this part was not built.
[image]
[image]

BreederBai, 2024-04-25:
@HTRamsey May I ask how to set the GST.FOUND in cmakelists? The issue I encountered with no RTSP seems to be due to the fact that GST_FOUND is false
[image]

BreederBai, 2024-04-25:
@HTRamsey It seems that there is an issue with this cmakelists file. I will build it in Windows 11
[image]

HTRamsey, 2024-04-25:
So if the gstreamer download is in the root folder it will pass that part successfully. It's working correctly on the CI build.

BreederBai, 2024-04-26:
> 因此，如果 gstreamer 下载位于根文件夹中，它将成功通过该部分。它在 CI 构建上工作正常。

My build and operation are also normal, but there is no gstreamer related video decoding option in the video options.During the construction process, it was found that the gstreamer folder was found, but GST was not found.Below is my error message when deleting the gstreamer folder.
[image]
Below is an error message when I added the gstreamer folder.
[image]
By comparison, it was found that the gstreamer folder can be found, indicating that there may have been some issues while executing pkg_check_modules. I have no knowledge of cmake syntax, so I cannot determine what the problem is. My CMake version is 3.27.7

BreederBai, 2024-04-26:
@HTRamsey I encountered the same problem when building QGC for the Windows platform. The problem is now quite clear, and it is highly likely that cmake did not include Gstreamer.

HTRamsey, 2024-04-26:
This is what I get when I do it like CI does, 
[image]
I guess you can just wait until I make so that CMake automatically downloads gstreamer for you

BreederBai, 2024-04-26:
> 这就是我像 CI 那样做时得到的结果， 我想你可以等到我 make 后，CMake 会自动为你下载 gstreamer ![图像](https://private-user-images.githubusercontent.com/68555040/325835320-8ea25e89-68f2-47e1-ad27-e529f85fd30c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxMDE3MzMsIm5iZiI6MTcxNDEwMTQzMywicGF0aCI6Ii82ODU1NTA0MC8zMjU4MzUzMjAtOGVhMjVlODktNjhmMi00N2UxLWFkMjctZTUyOWY4NWZkMzBjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI2VDAzMTcxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZjNzk5OTg0ZDM0YjQ4OWZmZTEyZTAzYzU4MGRmOTcyZjA0ODhjNTRkMDQ1Y2Q1ZTU1ZjdlMzUyNGNmOWY5ODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.mGpVv297Jf6tp1diElGNnei6xrDka0gRCWXJwg87IcM)

The version I downloaded is also 1.18.5, is it because I built it on Windows? I'll give it a try in Ubuntu and see what the results are

BreederBai, 2024-04-26:
@HTRamsey I successfully built it on Ubuntu 2004 and ran QGC on Android. When building, there was no "GST not found" error message, and the program also has gstreamer related functions after running. Through comparison, it can be determined that there are errors in Android building under Windows. This also explains why CI can run successfully.

HTRamsey, 2024-04-26:
Okay, I will try it out on Windows and see what the issue is

BreederBai, 2024-04-27:
> Okay, I will try it out on Windows and see what the issue is

Okay, thank you very much.

bobby-walle, 2024-05-13:
@BreederBai 是否方便联系您，我在编译是遇到一些问题

BreederBai, 2024-05-14:
> @BreederBai 是否方便联系您，我在编译是遇到一些问题

我的邮箱songshu_bai@outlook.com

closed 2025-05-03 by HTRamsey